neka-nat/cupoch #1: Building from source error with CUDA & eigen3

devshank3, 2020-02-23:
Hello @neka-nat,

A huge thanks for Cupoch

But im facing issue while building from source 

system config

Nvidia geforce 1650
cuda toolkit 10.2
ubuntu 19.10
inbuilt Eigen : eigen 3.3.7

I tried with both eigen3 3rdparty and inbuilt version using 
`cmake -DBUILD_EIGEN3=OFF`
and 
`cmake -DBUILD_EIGEN3=ON`

the make `make install-pip-package -j`  is getting killed 

few error snippets are 

`/home/shankar/3D projects/cupoch/cupoch/src/cupoch/geometry/boundingvolume.cu(26): error: identifier "Eigen::MatrixBase< ::Eigen::Matrix<float, (int)3, (int)3, (int)0, (int)3, (int)3> > ::determinant const" is undefined in device code
`
`36 errors detected in the compilation of "/tmp/tmpxft_00007184_00000000-6_boundingvolume.cpp1.ii".
CMake Error at cupoch_geometry_generated_boundingvolume.cu.o.cmake:280 (message):
  Error generating file /home/shankar/3D
  projects/cupoch/cupoch/build/src/cupoch/geometry/CMakeFiles/cupoch_geometry.dir//./cupoch_geometry_generated_boundingvolume.cu.o
`
`make[1]: *** [CMakeFiles/Makefile2:755: src/cupoch/geometry/CMakeFiles/cupoch_geometry.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
`

neka-nat, 2020-02-23:
Hi @devshank3 ,

Thank you for reporting.
Compilation may not pass with version 3.3.7 of eigen.
When using third_party, the version should be 3.3.90.
Please try the following:
* Remove CMakeCache.txt
* Remove eigen3 installed on the system

devshank3, 2020-02-23:
Now tried with  eigen3 installed was 3.3.90 

still got killed 

[image]

neka-nat, 2020-02-23:
"Killed" messages are like someone has sent Kill signals.
Suspect Kill signal from a system such as OOMKiller.

devshank3, 2020-02-23:
Sure i will try and let u know for  **building from source**

Mean while i installed Cupoch in my system via pip 

but few errors 

[image]
[image]
[image]

mainly cuda mem alloc

neka-nat, 2020-02-23:
I have updated the version of the pip package (v0.0.3).
Please install it again and try it.

devshank3, 2020-02-23:
works great thank you

neka-nat, 2020-02-24:
Thank you!
It seems that there was still a fatal defect.
I have upgraded the version to 0.0.4, so please try again.

closed 2020-02-24 by neka-nat