neka-nat/cupoch #2: cupoch.cupoch.device_vector_vector3f can not convert to numpy array?

longwoo, 2020-03-27:
Hello, 

Your Cupoch is really a useful tools.

when I run the example `clustering.py` , it gives me 
`Traceback (most recent call last):
  File "my_clustering.py", line 36, in <module>
    print("%s has %d points" % (pcl_name, np.asarray(pcl.points).shape[0]))
IndexError: tuple index out of range`

It seems that cupoch.cupoch.device_vector_vector3f can not convert to numpy array?

By the way my Os is Jectson Pack 4.2 with ubuntu18.04, and Cupoch is build from source with python2.7 binding.

Could you give me some advice?

Thank you.

neka-nat, 2020-03-27:
Hi @longwoo ,

Thank you for reporting.
Please try the conversion in the following way.

```
np.asarray(pcd.points.cpu())
```

And DBSCAN clustering may not work. I am currently debugging.

longwoo, 2020-03-27:
@neka-nat 
Thank you for your quick answer. It solves my problem.

And I want to make sure that `pcd.crop` is not support `OrientedBoundingBox` type yet?  Since it gives me 
```
    pc_ground = pcd.crop(box_ground)
TypeError: crop(): incompatible function arguments. The following argument types are supported:
    1. (self: cupoch.cupoch.geometry.PointCloud, bounding_box: cupoch::geometry::AxisAlignedBoundingBox) -> cupoch.cupoch.geometry.PointCloud

Invoked with: geometry::PointCloud with 139003 points., geometry::OrientedBoundingBox

```

So I need to use `AxisAlignedBoundingBox` as input instead of `OrientedBoundingBox`.

neka-nat, 2020-03-27:
I added crop function that takes OrientedBoundingBox.
You can use it by fetching the latest "master" branch and compiling it.

closed 2020-04-02 by neka-nat

pbdahzou, 2023-06-21:
> Hi @longwoo ,
> 
> Thank you for reporting. Please try the conversion in the following way.
> 
> ```
> np.asarray(pcd.points.cpu())
> ```
> 
> And DBSCAN clustering may not work. I am currently debugging.

Can DBSCAN clustering work correctly ?

pbdahzou, 2023-06-21:
fragment has 196133 points
[2023-06-21 21:12:18.968] [debug] Precompute Neighbours
Precompute Neighbours[========================================] 100%
fragment has 196133 clusters
fragment_o3d has 196133 points