neka-nat/cupoch #3: Visual Studio 2019 and Cuda 10.2 compile error

Wavelet303, 2020-05-06:
Hi, 
First of all, thanks for your amazing work. I have got several isues whne compile the project with visual studio. My setup is:

- Visual studio 2019
- Cuda 10.2
- Graphic card rtx 2060

In cupoch_registration my compliation error is:

**- Error		calling a __host__ function("isinf<float> ") from a __device__ function("thrust::cuda_cub::cub::AgentReduce< ::thrust::cuda_cub::cub::AgentReducePolicy<(int)256, (int)20, (int)2, ( ::thrust::cuda_cub::cub::BlockReduceAlgorithm)2, ( ::thrust::cuda_cub::cub::CacheLoadModifier)0> ,  ::thrust::cuda_cub::transform_input_iterator_t<float,  ::thrust::counting_iterator<int,  ::thrust::use_default,  ::thrust::use_default,  ::thrust::use_default> ,  ::_NV_ANON_NAMESPACE::extact_knn_distance_functor> , float *, int,  ::thrust::plus<float> > ::ConsumeTile<(int)1, (int)0> ") is not allowed	cupoch_registration	C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.2\include\thrust\iterator\iterator_facade.h	171	

- Error		identifier "isinf<float> " is undefined in device code	cupoch_registration	C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.2\include\thrust\iterator\iterator_facade.h	171	

- Error		calling a __host__ function("isinf<float> ") from a __device__ function("thrust::cuda_cub::cub::AgentReduce< ::thrust::cuda_cub::cub::AgentReducePolicy<(int)256, (int)20, (int)2, ( ::thrust::cuda_cub::cub::BlockReduceAlgorithm)2, ( ::thrust::cuda_cub::cub::CacheLoadModifier)0> ,  ::thrust::cuda_cub::transform_input_iterator_t<float,  ::thrust::counting_iterator<int,  ::thrust::use_default,  ::thrust::use_default,  ::thrust::use_default> ,  ::_NV_ANON_NAMESPACE::extact_knn_distance_functor> , float *, int,  ::thrust::plus<float> > ::ConsumeTile<(int)1, (int)0> ") is not allowed	cupoch_registration	C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.2\include\thrust\iterator\iterator_facade.h	171	

- Error		identifier "isinf<float> " is undefined in device code	cupoch_registration	C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.2\include\thrust\iterator\iterator_facade.h	171	

- Error		calling a __host__ function("isinf<float> ") from a __device__ function("thrust::cuda_cub::cub::AgentReduce< ::thrust::cuda_cub::cub::AgentReducePolicy<(int)256, (int)20, (int)2, ( ::thrust::cuda_cub::cub::BlockReduceAlgorithm)2, ( ::thrust::cuda_cub::cub::CacheLoadModifier)0> ,  ::thrust::cuda_cub::transform_input_iterator_t<float,  ::thrust::counting_iterator<int,  ::thrust::use_default,  ::thrust::use_default,  ::thrust::use_default> ,  ::_NV_ANON_NAMESPACE::extact_knn_distance_functor> , float *, int,  ::thrust::plus<float> > ::ConsumeTile<(int)0, (int)0> ") is not allowed	cupoch_registration	C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.2\include\thrust\iterator\iterator_facade.h	171	

- Error		identifier "isinf<float> " is undefined in device code	cupoch_registration	C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.2\include\thrust\iterator\iterator_facade.h	171	

- Error		calling a __host__ function("isinf<float> ") from a __device__ function("thrust::cuda_cub::cub::AgentReduce< ::thrust::cuda_cub::cub::AgentReducePolicy<(int)256, (int)20, (int)2, ( ::thrust::cuda_cub::cub::BlockReduceAlgorithm)2, ( ::thrust::cuda_cub::cub::CacheLoadModifier)0> ,  ::thrust::cuda_cub::transform_input_iterator_t<float,  ::thrust::counting_iterator<int,  ::thrust::use_default,  ::thrust::use_default,  ::thrust::use_default> ,  ::_NV_ANON_NAMESPACE::extact_knn_distance_functor> , float *, int,  ::thrust::plus<float> > ::ConsumeTile<(int)0, (int)0> ") is not allowed	cupoch_registration	C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.2\include\thrust\iterator\iterator_facade.h	171	

- Error		identifier "isinf<float> " is undefined in device code	cupoch_registration	C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.2\include\thrust\iterator\iterator_facade.h	171	

- Error		calling a __host__ function("isinf<float> ") from a __device__ function("thrust::cuda_cub::cub::LoadDirectStriped<(int)256, float, (int)20,  ::thrust::cuda_cub::transform_input_iterator_t<float,  ::thrust::counting_iterator<int,  ::thrust::use_default,  ::thrust::use_default,  ::thrust::use_default> ,  ::_NV_ANON_NAMESPACE::extact_knn_distance_functor> > ") is not allowed	cupoch_registration	C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.2\include\thrust\iterator\iterator_facade.h	171	

- Error		identifier "isinf<float> " is undefined in device code	cupoch_registration	C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.2\include\thrust\iterator\iterator_facade.h	171**	

on the other hand at cupoch_utility i have other compile error:

**- Error		asm operand type size(8) does not match type/size implied by constraint 'r'	cupoch_utility	E:\wavelet\cupoch\third_party\eigen\Eigen\src\Core\GenericPacketMath.h	429	

- Error		asm operand type size(8) does not match type/size implied by constraint 'r'	cupoch_utility	E:\wavelet\cupoch\third_party\eigen\Eigen\src\Core\GenericPacketMath.h	429**	

how can i solve this errors? Thanks for all ;)

neka-nat, 2020-05-06:
Hi @Wavelet303 ,

Thank you for your reporting.
This library currently don't support Windows.
I'm thinking supporting MINGW on windows at first and next MVSC.

neka-nat, 2020-07-08:
Hi @Wavelet303 ,

The following branch has been successfully compiled in msvc.
https://github.com/neka-nat/cupoch/tree/msvc
Please check the issue, #8 .

neka-nat, 2020-07-09:
Merge into the newer issue #8 .

closed 2020-07-09 by neka-nat