neka-nat/cupoch #4: ScalableTSDF

VladMVLX, 2020-06-06:
Hello,

First of all I want to say thank you for the amazing work you doing and sharing.

Do you have any plans to integrate open3d implementation of ScalableTSDF or any analogue of it?

Or will it be working with cupoch if I will take scalable tsdf class from cuda branch of open3d and compile it with cupoch (fixing namespaces and names)?

Thanks.

neka-nat, 2020-06-07:
Hi @VladMVLX ,

Thank you for your feedback.
The implementation of scarable tsdf will be done in the future.
It need to be modified for CUDA based on the open3d source.

neka-nat, 2020-07-28:
I implemented TSDF generation and point cloud extraction
The implementation of mesh extraction is not yet done.
Also, as you can see in #12 , it's not memory efficient.