neka-nat/cupoch #5: Question about collision detection

baoyufuyou, 2020-06-17:
Hi, Thanks a lot for your amazing work. Could I ask, did you test the collision checking part?
Currently I wanna test the collision detection given two point clouds or meshes and corresponding translation matrixes. If you have the test code for it, that will be helpful. Thanks again. ;)

Best regard,
lei

neka-nat, 2020-06-17:
Hi @baoyufuyou ,

Thanks for the feedback.
The collision module has some APIs available, but it has not been fully tested yet.
I plan to add more python examples in the future.
If you try it out and find a problem, please send me issues or pull requests.

neka-nat, 2020-11-07:
I added some python examples for collision detection.

https://github.com/neka-nat/cupoch/blob/master/examples/python/robotics/collision_linset_voxel.py
https://github.com/neka-nat/cupoch/blob/master/examples/python/robotics/collision_voxelgrid.py