neka-nat/cupoch #8: When will windows be supported?

caibf, 2020-07-07:
Hi everyone,

I am so greatly to hear that the libray was able to process 3D point cloud data using CUDA. Furthermore, _cupoch_ is based on the functionality of Open3D. 

From this issue: [https://github.com/neka-nat/cupoch/issues/3](url), I know that the libray didn't support either MINGW or MSVC on windows.

However, my app developed with **MSVC** and ran on windows. Up to now, I use Open3D to implement my requirements on CPU yet. I'm very anxious to speed up my app by running on GPU.

When will windows be supported?

VladMVLX, 2020-07-07:
Hi caibf,

You can actually compile cupoch for windows using CMAKE and MSVC and cuda toolkit installed, not from scratch as it is require some manual work and adjustments, but I was able to compile it and use it in my windows based project.

There is some huge problems with rmm so try to avoid building with it, everything else works just fine but needs some minor manual adjustments in code and in build scrips.

neka-nat, 2020-07-08:
Hi,

Thanks for the feedback.
I am now trying to see if I can compile with MSVC without rmm.
https://github.com/neka-nat/cupoch/tree/msvc
After a week or so, I'll know if I can compile it.

caibf, 2020-07-08:
Hi VladMVLX,

Wonderful news for me! Could you give me the detailed guide to me about how to make some minor adjustments in code and in build scripts?

neka-nat, 2020-07-08:
Hi,

The following branch has been successfully compiled in msvc.
https://github.com/neka-nat/cupoch/tree/msvc
The environment is as follows.

* Windows10 64bit
* Visual Studio 2019
* python3.7
* CUDA10.1

I have checked import, but have not tested it.
You can also install with pip.

```
pip install cupoch
```

neka-nat, 2020-07-24:
Hi,

You can compile it with msvc in the latest master branch.

* Windows10 64bit
* Visual Studio 2019
* python3.7
* CUDA10.2

You can also install it with pip.

closed 2020-07-24 by neka-nat

caibf, 2020-09-07:
Hey @neka-nat ,

As far as I know, the third party of rmm only supports on linux(with Python versions 3.6 or 3.7). How did you solve this problem that compiled with windows?

neka-nat, 2020-09-07:
Hi @caibf ,

When compiling on Windows, rmm is disabled.
The default allocator of thrust is used.

caibf, 2020-09-08:
OK. Thanks @neka-nat. _Thrust_ is a library that focus on parallel algorithms and data structures, but _RMM_ is a pool allocator to make CUDA device memory allocation / deallocation faster and asynchronous. _Thrust_ has the high-level interface for GPU programming, so you don't need to take care of memory access and allocation. 

Therefore, what's the performance different between _thrust_ and _rmm_. Whether I can say, _rmm_ has a good performance than _thrust_ on linux in the same hareware environment.

neka-nat, 2020-09-08:
I feel that using the RMM pool allocator would be slightly faster.
No big changes.