neka-nat/cupoch #9: Raycasting

destrooooyer, 2020-07-23:
Hi,
Thanks so much for your work.
Do you have any plan implementing the ray casting algorithm to get a depth map / point cloud from the TSDF volume for a given camera pose?

Best regards

neka-nat, 2020-07-23:
Hi @destrooooyer ,

Thank you for the question.
Is what you're asking about the following?

https://github.com/neka-nat/cupoch/blob/master/examples/python/advanced/rgbd_integration.py
https://github.com/neka-nat/cupoch/blob/master/examples/python/advanced/rgbd_integration_uniform.py

destrooooyer, 2020-07-23:
Hi @neka-nat 
No I mean the ray casting method in KinectFusion. In this algorithm you cast a lot of rays from the camera pose to the TSDF volume, which gives an estimation of the surface.

Here is a simple introduction to the algorithm. (Ray casting on page 10)
https://vision.in.tum.de/_media/teaching/ss2015/gpucourse_ss2015/kinectfusion.pdf 

And here is an implementation of ray casting
https://github.com/chrdiller/KinectFusionLib/blob/master/src/cuda/surface_prediction.cu

neka-nat, 2020-07-23:
Thank you for the references.
I hadn't planned to implement it, but it's very interesting.
Currently, I'm working on other issues as well, so I can't say when I'll be able to implement it,
but I put it on the list of future features.

neka-nat, 2021-01-01:
I added `Raycast` function to `UniformTSDFVolume` .
https://github.com/neka-nat/cupoch/blob/master/src/cupoch/integration/uniform_tsdfvolume.h#L77
I also experimentally implemented Kinect Fusion, so try it out.
https://github.com/neka-nat/cupoch/tree/master/src/cupoch/kinfu

closed 2021-01-02 by neka-nat