neka-nat/cupoch #10: Is it possible to use Open3D and Cupoch together?

1939938853, 2020-07-26:
Hello, 

I tried to use Open3d and Cupoch together side-by-side for comparison. But got something redefined error. Here is my code

`
 auto pcd_gpu = cupoch::io::CreatePointCloudFromFile("pcd1.pcd");
 auto pcd = open3d::io::CreatePointCloudFromFile("pcd1.pcd");
`

Here is the error messages

[image]

Anyway to solve this issue?

Thank you!

neka-nat, 2020-07-26:
Hi @1939938853 ,

Thank you for the reporting.
It sounds like a problem with fmt and tinyobjloader libraries.
I think the following issues are related to the tinyobjeloader one.
https://github.com/tinyobjloader/tinyobjloader/issues/171
I didn't find any related issues with fmt.

1939938853, 2020-07-28:
It looks that we need to manually modify the header files of both fmt and tinyobjloader.

By the way, what is the suggestion for best covert Open3D 3D data to Cupoch and vice versa. static_cast or via the eigenvector?

Thank you!

neka-nat, 2020-07-28:
The open3d point cloud is a std::vector.
By contrast, cupoch is a thrust::device_vector.
It can be copied by the assignment operator.
https://stackoverflow.com/questions/43982841/how-to-copy-host-vector-to-device-vector-by-thrust

```
open3d::geometry::PointCloud pcd_cpu;
// Initialize pcd_cpu...
cupoch::geometry::PointCloud pcd_gpu;
pcd_gpu.points_ = pcd_cpu.points_;
```

closed 2020-08-10 by neka-nat

1939938853, 2020-11-06:
Want to use Cupoch together with Open3D but facing the problem

Cupoch use CMAKE_CXX_FLAG MD/MDd while Open3D use MT/MTd. Tried several times to change Cupoch cmake option to MD in building Cupoch, but without success -- no change happend.