neka-nat/cupoch #11: Is Cupoch Windows MSVC GPU ready to work?

1939938853, 2020-07-26:
Hello, 

I tried to test Cupoch on Windows 10.

1. Build with VS2019 --> no problem. 
2. Test normal calculation with the same 18519 points PCD --> Open3D  needs 10ms, Cupoch needs 52 ms    

Here is the code 

```
pcd->EstimateNormals(open3d::geometry::KDTreeSearchParamHybrid(radius_normal, MAX_NN));
pcd_gpu->EstimateNormals(cupoch::geometry::KDTreeSearchParamHybrid(radius_normal, MAX_NN));
```

I guess that the calculation was not run on GPU. But how to make sure it is run on GPU? I already had this line in my 

 ` cupoch::utility::InitializeAllocator();`

neka-nat, 2020-07-26:
Hi @1939938853 ,

Thanks you for trying!
The difference between the windows version and the linux version is that the rmm library,
which accelerates memory allocation, is disabled in the windows version.

On linux, with rmm's memory allocation disabled, about 10,000 points, open3d with OpenMP enabled,
the results are below.

[image]

At about 10,000 points, you can see that there is almost no difference in speed between CPU and GPU.

The next comparison is for 100,000 points, other conditions being the same.

[image]

A 100,000 point calculation can show the effectiveness of the GPU.
Especially estimate_normals are hard to make a difference.

1939938853, 2020-07-26:
Hello Neka-Nat,

Thank you for your explanation.  

On the first figure,  do you mean "On linux, with rmm's memory allocation ENABLED" ?

I saw you use STDGPU. Can STDGPU on windows plays the same role as RMM on Linux for acceleration? What is the GPU acceleration situation of Cupoch on Windows now?

neka-nat, 2020-07-27:
Thank you for replying.
All of the above calculations have rmm disabled.
On linux, you can choose to enable or disable rmm.
On windows, rmm is disabled.
STDGPU has no effect on the speed.

1939938853, 2020-07-27:
Hello, 

Here is my test
 
1.  For 18519 points cloud，normal estimation (KDTreeSearchParamHybrid)， Open3D 10 ms, Cupoch 52 ms.
2.  For 592005 points cloud， normal estimation (KDTreeSearchParamHybrid)， Open3D 2540 ms, Cupoch 6279 ms.
3.  for 198835 points cloud (cupoch data) ,  ICP registration， Open3D 1499 ms, Cupoch 221 ms.

It looks that for ICP registration, the Cupoch GPU speedup (acceleration) is quite effective. But for normal estimation, why Cupoch is much slower than Open3D?

Any suggestion?

Thank you!

neka-nat, 2020-07-28:
Hi,

Thank you for trying.
The max_nn parameter used for estimate_normals affects the calculation speed.
In my experiment above, I set max_nn to 30, but if I set max_nn to 100, the CPU is faster.
You can improve the speed by setting max_nn to a smaller value.

1939938853, 2020-07-28:
Yes. When I set max_nn to 30,  with 592005 points cloud, the Cupoch normal estimation duration now is  231 ms.

closed 2020-07-28 by neka-nat