neka-nat/cupoch #15: version `GLIBC_2.29' not found

flugenheimer, 2020-08-20:
Currently trying to test this library out in google colab, however I am running into an issue where something seems to be looking for libm.se.6 version BLIBC_2.29. 

Running ldd --version shows me the google colab server is running 2.27:
`ldd (Ubuntu GLIBC 2.27-3ubuntu1) 2.27`

From what i can see 2.27 is standard for ubuntu 1.18. for this library it is stated that ubuntu 1.18 is supported

Supported platforms
Ubuntu 18.04

Why does it not work with GLIBC 2.27 ? 

```
/usr/local/lib/python3.6/dist-packages/cupoch/__init__.py in <module>()
----> 1 from .cupoch import * # py2 py3 compatible
      2 initialize_allocator()
      3 
      4 __version__ = '0.1.2.0'

ImportError: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /usr/local/lib/python3.6/dist-packages/cupoch/cupoch.cpython-36m-x86_64-linux-gnu.so)
```

neka-nat, 2020-08-20:
Hi @flugenheimer ,

Thanks you for reporting the issue!
The problem is that the latest package was compiled in Ubuntu 20.04.
I just updated python 3.6 package to v0.1.3.1 by compiling it at 18.04.
Please try updating it with pip.

```
pip install cupoch -U
```

flugenheimer, 2020-08-21:
> Hi @flugenheimer ,
> 
> Thanks you for reporting the issue!
> The problem is that the latest package was compiled in Ubuntu 20.04.
> I just updated python 3.6 package to v0.1.3.1 by compiling it at 18.04.
> Please try updating it with pip.
> 
> ```
> pip install cupoch -U
> ```

Hi neka-nat, it works locally on my machine now, however google colab session crashes after trying to read a .ply file
```
import numpy as np
import time 
import cupoch as cph

source_gpu = cph.io.read_point_cloud("1.ply")
```

closed 2020-08-24 by neka-nat