neka-nat/cupoch #17: Direct zero-copy from cupy array

Shipborn, 2020-08-28:
I failed to find in the documentation or anywhere in the code comments a way of directly using cupy arrays that I already have loaded on the device?

Right now I am running custom kernels using numba and cupy and it's working like a charm. However when I use cupoch I first need to copy to host and then copy it back to the device using cupoch. This currently takes 30ms on an algorithm that takes a total of 30ms.... so 50% is lost on wasted copying.

Right now I have this workflow
`cloud = cupy.asarray(source_data)`
`*Custom calculations*`
`cloud_host = cupy.asnumpy(cloud)`
`gpu = cph.geometry.PointCloud()`
`gpu.points = cph.utility.Vector3fVector(cloud_host)`

Im sure there is a faster way :)

neka-nat, 2020-08-28:
Hi @Shipborn ,

Thank you for the feedback.
You can use dlpack to interact with data in cupy.

```py
cloud = cupy.asarray(source_data)
dltensor = cloud.toDlpack()
gpu = cph.geometry.PointCloud()
gpu.from_points_dlpack(dltensor)
```

closed 2020-08-31 by neka-nat

dHofmeister, 2020-09-02:
Thank you @neka-nat,

for reference for others:
It currently says in the variable description "Use `numpy.asarray()` to copy to host."
Which is identical to the open3D description, implying:
`host_array = np.asarray(gpu.points)`

but the actual code is:
`host_array = np.asarray(gpu.points.cpu())`
 This is not following the open3D API. not a critique, just a heads-up :) 

just for future people to see.

Greetings
Deniz

neka-nat, 2020-09-02:
Hi @dHofmeister ,

Thank you for the report!
I've corrected the wording and pushed it to master.
Thanks.