neka-nat/cupoch #19: How to create a mask for Occupancy Grid free area?

toshi2k2, 2020-09-04:
for the `geometry.OccupancyGrid()` method, how do we manually set the free area?
Is there any documentation for this method?

neka-nat, 2020-09-04:
Thanks for the feedback.
Do you mean the following function?
```
occupancygrid.set_free_voxels(voxels)
```
It is not currently implemented.

toshi2k2, 2020-09-04:
Yes. 
Is there any documentation for the `OccupancyGrid()` method? I'm trying to figure out from the examples as to how return an occupancy map given a point cloud or rgb and depth image.

neka-nat, 2020-09-05:
I haven't started work on the documentation yet.
Please refer to the example of OccupancyGrid.
https://github.com/neka-nat/cupoch/blob/master/examples/python/realsense/realsense_occupancygrid_viewer.py
You can get the voxel grid itself with the `voxels` property, but only for occupied voxels.

I will continue to add the following features to OccupancyGrid:
* Conversion between geometries, VoxelGrid <-> OccupancyGrid, DistanceTransform <-> OccupancyGrid, ...
* Adding examples with collision detection
* Access function to grid other than `insert`
If you have any feature requests, please let me know. Pull requests are also welcome.

neka-nat, 2020-10-06:
I added some functions for OccupancyGrid.

* OccupancyGrid::SetFreeArea
* DistanceTransformation::CreateFromOccupancyGrid
* VoxelGrid::CreateFromOccupancyGrid

closed 2020-10-07 by neka-nat