neka-nat/cupoch #21: Is it possible to use marching cubes mesh extraction for xyz point cloud?

1939938853, 2020-09-22:
Hi,

Instead of [RGBD](https://github.com/neka-nat/cupoch/tree/master/src/cupoch/integration), just wonder it is possible to use marching cubes mesh extraction for xyz point cloud?

Thanks

neka-nat, 2020-09-22:
Hi @1939938853 ,

Thank you for your feedback.
Do you want to generate a mesh from a point cloud?
BPA and other methods seem to be better than marching cubes.

https://towardsdatascience.com/5-step-guide-to-generate-3d-meshes-from-point-clouds-with-python-36bad397d8ba

These are not implemented yet.

1939938853, 2020-09-22:
I have tried the open3d Ball-Pivoting and Poisson. Plus normal estimation and normal rearrangement, it is difficult to achieve desired performance and rendering effect.

The Marching Cubes looks promising. But may need to  turn a point cloud into a (T)SDF volume?

1939938853, 2020-09-22:
From [here](https://github.com/PointCloudLibrary/pcl/blob/master/tools/marching_cubes_reconstruction.cpp), it looks that PCL can use Marching Cubes to create mesh?

neka-nat, 2020-09-23:
Thanks for the reference.
It looks like PCL voxelizes the point cloud and applies Marching Cubes.
I add it to a future feature.

1939938853, 2020-09-23:
Marching Cubes based surface reconstruction seems promising in the aspect of real-time related performance.

-  Can avoid normal estimation? Normal estimation is not so "cheap", also GPU acceleration is not so effective, especially when the point cloud is not so big. Furthermore to make all normal direction towards outside is not so "cheap" either.
-  From [Nvidia's ](https://www.nvidia.com/content/gtc-2010/pdfs/gtc10_session2020_hp5mc_dyken_ziegler.pdf) early presentation, it seems that Marching Cubes is parallelization friendly?

Surface reconstruction from point cloud, except meshing, perhaps the other better choice is the [surface splatting](https://github.com/sebastianlipponer/surface_splatting)? I guess that the Cupoch visualization of point cloud with normal has some kind of splatting? But it is not "water-tight". If  Cupoch  can also realize the surface splatting as shown in the this[ link]( https://github.com/sebastianlipponer/surface_splatting ) , that could be great. Even though surface splatting require per-calculated normal, but seems that it is not sensitive to the normal direction.

neka-nat, 2020-09-24:
Thank you for the some suggestions.
I didn't know about surface-splatting.
The marching cubes still sounds advantageous from the point of view of
not having to calculate the normals and the speed of the calculation.
I'll look into surface-splatting as well.
Thanks!

manhha1402, 2020-10-27:
Hi,
Are you planning to add the extraction of MarchingCubes from ScalableTSDFVolume? Since I think we can reconstruct larger area of the environment.

neka-nat, 2020-10-27:
Yes,
I are planning to add it.
Please wait for a while.

closed 2021-02-21 by 1939938853