neka-nat/cupoch #24: Use the C++ cupoch package directly

Shipborn, 2020-10-01:
Dear @neka-nat,

My main question is: I was wondering if it is possible to use the cupoch c++ library directly in my own projects? If so, how do I structure my Cmakelists.txt and what do i link to where?

-------------------------------------------------------------------------------------------------------------------------
WHAT I TRIED:
**Installing:**
```
git clone https://github.com/neka-nat/cupoch.git --recurse
cd cupoch
mkdir build
cd build
cmake ..; make install-pip-package -j
```
but I was hoping i could also add the command:
```
sudo make install
```
to place it in my `usr/....` to make it findable for my own package.

**Using:**
I tried importing the cupoch package in my Cmakelists.txt as:
```
#find_package(cupoch REQUIRED)
add_executable(
        my_prog
        src/program.cpp
        )

install(TARGETS my_prog DESTINATION lib/${PROJECT_NAME})

target_link_libraries(my_prog cupoch_geometry cupoch_utility cupoch_visualization cupoch_io fmt::fmt)
ament_target_dependencies(my_prog Ceres CUDA)
```

Yet this doesnt work as it cannot find cupoch.
```
CMake Error at CMakeLists.txt:26 (find_package):
  By not providing "Findcupoch.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "cupoch", but
  CMake did not find one.

  Could not find a package configuration file provided by "cupoch" with any
  of the following names:

    cupochConfig.cmake
    cupoch-config.cmake

  Add the installation prefix of "cupoch" to CMAKE_PREFIX_PATH or set
  "cupoch_DIR" to a directory containing one of the above files.  If "cupoch"
  provides a separate development package or SDK, be sure it has been
  installed.
```
if i simply comment the find_package(cupoch REQUIRED) command, then it fails to find the fmt libraries:

```
[ 14%] Linking CXX executable my_prog
/usr/bin/ld: /usr/local/lib/libcupoch_geometry.a(cupoch_geometry_generated_boundingvolume.cu.o): in function `void fmt::v6::internal::basic_writer<fmt::v6::buffer_range<char> >::write_fp<float, true>(float, fmt::v6::basic_format_specs<char> const&)':
/home/shipborn/Documents/dashing/main/ros2/src/external/cupoch/third_party/fmt/include/fmt/format.h:2855: undefined reference to `bool fmt::v6::internal::grisu_format<double, 0>(double, fmt::v6::internal::buffer<char>&, int, unsigned int, int&)'
/usr/bin/ld: /usr/local/lib/libcupoch_geometry.a(cupoch_geometry_generated_boundingvolume.cu.o): in function `void fmt::v6::internal::basic_writer<fmt::v6::buffer_range<char> >::write_fp<double, true>(double, fmt::v6::basic_format_specs<char> const&)':
/home/shipborn/Documents/dashing/main/ros2/src/external/cupoch/third_party/fmt/include/fmt/format.h:2855: undefined reference to `bool fmt::v6::internal::grisu_format<double, 0>(double, fmt::v6::internal::buffer<char>&, int, unsigned int, int&)'
/usr/bin/ld: /usr/local/lib/libcupoch_geometry.a(cupoch_geometry_generated_kdtree_flann.cu.o): in function `flann::KDTreeCuda3dIndex<flann::L2<float> >::~KDTreeCuda3dIndex()':
/home/shipborn/Documents/dashing/main/ros2/src/external/cupoch/third_party/flann/algorithms/kdtree_cuda_3d_index.h:111: undefined reference to `flann::KDTreeCuda3dIndex<flann::L2<float> >::clearGpuBuffers()'
/usr/bin/ld: /usr/local/lib/libcupoch_geometry.a(cupoch_geometry_generated_kdtree_flann.cu.o): in function `flann::KDTreeCuda3dIndex<flann::L2<float> >::buildIndex()':
/home/shipborn/Documents/dashing/main/ros2/src/external/cupoch/third_party/flann/algorithms/kdtree_cuda_3d_index.h:138: undefined reference to `flann::KDTreeCuda3dIndex<flann::L2<float> >::uploadTreeToGpu()'
/usr/bin/ld: /usr/local/lib/libcupoch_geometry.a(cupoch_geometry_generated_kdtree_flann.cu.o): in function `flann::KDTreeCuda3dIndex<flann::L2<float> >::knnSearch(flann::Matrix<float> const&, flann::Matrix<int>&, flann::Matrix<float>&, unsigned long, flann::SearchParams const&) const':
/home/shipborn/Documents/dashing/main/ros2/src/external/cupoch/third_party/flann/algorithms/kdtree_cuda_3d_index.h:184: undefined reference to `flann::KDTreeCuda3dIndex<flann::L2<float> >::knnSearchGpu(flann::Matrix<float> const&, flann::Matrix<int>&, flann::Matrix<float>&, unsigned long, flann::SearchParams const&) const'
/usr/bin/ld: /usr/local/lib/libcupoch_geometry.a(cupoch_geometry_generated_kdtree_flann.cu.o): in function `flann::KDTreeCuda3dIndex<flann::L2<float> >::radiusSearch(flann::Matrix<float> const&, flann::Matrix<int>&, flann::Matrix<float>&, float, flann::SearchParams const&) const':
/home/shipborn/Documents/dashing/main/ros2/src/external/cupoch/third_party/flann/algorithms/kdtree_cuda_3d_index.h:201: undefined reference to `flann::KDTreeCuda3dIndex<flann::L2<float> >::radiusSearchGpu(flann::Matrix<float> const&, flann::Matrix<int>&, flann::Matrix<float>&, float, flann::SearchParams const&) const'
collect2: error: ld returned 1 exit status
make[3]: *** [CMakeFiles/pose_graph.dir/build.make:302: pose_graph] Error 1
make[2]: *** [CMakeFiles/Makefile2:152: CMakeFiles/pose_graph.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:159: CMakeFiles/pose_graph.dir/rule] Error 2
make: *** [Makefile:221: pose_graph] Error 2
```

Shipborn, 2020-10-02:
So i've managed to piece together that what was missing was:

```
link_directories("path__to__cupoch__:  /cupoch/build/lib")

find_package(fmt REQUIRED)
find_package(flann REQUIRED)

install(TARGETS my_prog DESTINATION lib/${PROJECT_NAME})

target_link_libraries(my_prog
        cupoch_geometry
        cupoch_utility
        cupoch_registration
        cupoch_visualization
        fmt
        flann_cuda_s
        )

```

-----------------------------------------

However, now I am running into an issue with memory allocation. 

given the function that converts ROS2 sensor_msgs::msgs::Pointcloud2 Messages to cupoch::geometry::Pointcloud (I wrote this, but i think you also have such a function hidden somewhere? ) :

```
cupoch::geometry::PointCloud poseGraph::prepare_pointcloud(const sensor_msgs::msg::PointCloud2 &msg){
    thrust::host_vector<Eigen::Vector3f> source_cloud(msg.width);
    for (unsigned long j = 0; j < msg.data.size(); j = j + 16) { // Assumes XYZI content
        Eigen::Vector3f z;
        for (unsigned int k = 0; k < 3; k++) // Assumes 4 steps of int8 per float ( Skips the I of XYZI component)
        {
            u_char v[] = {msg.data[j + k * 4 + 0],
                          msg.data[j + k * 4 + 1],
                          msg.data[j + k * 4 + 2],
                          msg.data[j + k * 4 + 3]};
            float w;
            memcpy(&w, &v, sizeof(v));
            z[k] = w;
        }
        source_cloud[j/16];
    }
    return cupoch::geometry::PointCloud(source_cloud);
}
```

now using:

```
void poseGraph::query_icp(const unsigned long idx_a,
                          const unsigned long idx_b,
                          const Vertex &vertex) {

    auto pointcloud_b = this->pointcloud_database[idx_b];  // type sensor_msgs::msg::Pointcloud2
    auto pointcloud_a = this->pointcloud_database[idx_a];

    cupoch::geometry::PointCloud pc_a = this->prepare_pointcloud(pointcloud_a);
    cupoch::geometry::PointCloud pc_b =  this->prepare_pointcloud(pointcloud_a);  // <---- Adding this line crashes the system.
/// If i remove the line, then such errors do not occur.
}
```

Error:
```
CUDA Error detected. cudaErrorInvalidValue invalid argument
pose_graph: /home/shipborn/Documents/dashing/main/ros2/src/external/cupoch/third_party/rmm/include/rmm/mr/device/cuda_memory_resource.hpp:83: virtual void rmm::mr::cuda_memory_resource::do_deallocate(void*, std::size_t, cudaStream_t): Assertion `status__ == cudaSuccess' failed.
Signal: SIGABRT (Aborted)
```

neka-nat, 2020-10-02:
Hi @Shipborn ,

Thank you for your feedback.
'cmake instal'l is currently not supported.
I plan to support it in the future.

For the error, please initialize the allocator when you run the program.

```cpp
cupoch::utility::InitializeAllocator();
```

Shipborn, 2020-10-05:
dear Neka, 

Thank you for your reply. 
Initializing the allocator does not solve the problem. It still throws me the same error.
The default InitializeAllocator() function does nothing?

The implementation is:
```
inline void InitializeAllocator(
        rmmAllocationMode_t mode = CudaDefaultAllocation,
        size_t initial_pool_size = 0,
        bool logging = false,
        const std::vector<int> &devices = {}) {}
```
The function I'm trying to call repeatedly is your RegistrationICP method. 

```
void poseGraph::query_icp(const unsigned long idx_a,
                          const unsigned long idx_b,
                          const Vertex &vertex) {

    sensor_msgs::msg::PointCloud2 pointcloud_b = this->pointcloud_database[idx_b];
    sensor_msgs::msg::PointCloud2 pointcloud_a = this->pointcloud_database[idx_a];

    ////-------------------- This point crashes at the second entry into this function.--------------
    thrust::host_vector<Eigen::Vector3f> source_cloud(pointcloud_a.width); // <------------------------------------------

    for (unsigned long j = 0; j < pointcloud_a.data.size(); j = j + 16) {
        Eigen::Vector3f z;
        for (unsigned int k = 0; k < 3; k++) // Assumes 4 steps of int8 per float
        {
            u_char v[] = {pointcloud_a.data[j + k * 4 + 0],
                          pointcloud_a.data[j + k * 4 + 1],
                          pointcloud_a.data[j + k * 4 + 2],
                          pointcloud_a.data[j + k * 4 + 3]};
            float w;
            memcpy(&w, &v, sizeof(v));
            z[k] = w;
        }
        source_cloud[j/16] = z;
    }
    auto source = std::make_shared<cupoch::geometry::PointCloud>(source_cloud);

    thrust::host_vector<Eigen::Vector3f> target_cloud(pointcloud_b.width);
    for (unsigned long j = 0; j < pointcloud_b.data.size(); j = j + 16) {
        Eigen::Vector3f y;
        for (unsigned int k = 0; k < 3; k++) // Assumes 4 steps of int8 per float
        {
            u_char v[] = {pointcloud_b.data[j + k * 4 + 0],
                          pointcloud_b.data[j + k * 4 + 1],
                          pointcloud_b.data[j + k * 4 + 2],
                          pointcloud_b.data[j + k * 4 + 3]};
            float w;
            memcpy(&w, &v, sizeof(v));
            y[k] = w;
        }
        target_cloud[j/16] = y;
    }

    auto target = std::make_shared<cupoch::geometry::PointCloud>(target_cloud);
    auto res = cupoch::registration::RegistrationICP(
                                                     *source,
                                                     *target,
                                                     0.20);
```
this works, but only the first loop. I am running this function multiple times and it always executes fine the first loop. but then crashes somewhere in the 2 - 4th loop. The error thrown is:
```
corrupted size vs. prev_size
Signal: SIGABRT (Aborted)
```
or:
```
free(): invalid pointer
Signal: SIGABRT (Aborted)
```

I do call your initializer in the main function:
```
int main(int argc, char **argv) {
    cupoch::utility::InitializeAllocator();
    rclcpp::init(argc, argv);

    rclcpp::executors::SingleThreadedExecutor executor;
    auto node = std::make_shared<poseGraph>();
    executor.add_node(node);
    executor.spin();
    rclcpp::shutdown();
    return 0;
}
```

I would like to know where the error comes from when the function runs for a second time? I assumed that the end of the function it is all properly freed and garbage collected?  

Greetings and thank you very much for your time.

neka-nat, 2020-10-06:
Thank you for the feedback!
Can I get the source code that I can run?
I'd like to debug it in my environment.

neka-nat, 2020-10-09:
If it's difficult to share the code, is it possible to use gdb or something similar to find out more about debugging information?

ZhenshengLee, 2021-01-18:
# how to use the cupoch c++ library
the following steps work in ubuntu1804
## install cupoch with 3rdparty libs
add some install command to  `third_party/CMakeLists.txt`
```
# Installation
install(TARGETS ${FLANN_LIBRARIES}
                ${fmt_LIBRARIES}
                ${rmm_LIBRARIES}
                ${liblzf_LIBRARIES}
                ${rply_LIBRARIES}
        EXPORT ${PROJECT_NAME}3rdTargets
        RUNTIME DESTINATION ${CUPOCH_INSTALL_BIN_DIR}
        LIBRARY DESTINATION ${CUPOCH_INSTALL_LIB_DIR}
        ARCHIVE DESTINATION ${CUPOCH_INSTALL_LIB_DIR}
)

install(DIRECTORY ${flann_INCLUDE_DIRS}
                    ${fmt_INCLUDE_DIRS}
                    ${cnmem_INCLUDE_DIRS}
                    ${rmm_INCLUDE_DIRS}
                    ${liblzf_INCLUDE_DIRS}
                    ${rply_INCLUDE_DIRS}
    DESTINATION ${CUPOCH_INSTALL_INCLUDE_DIR}/../
)
```
I will not use the python interface so
```
cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/opt/cupoch/cupoch -DBUILD_UNIT_TESTS=OFF -DBUILD_PYBIND11=OFF -DBUILD_PYTHON_MODULE=OFF
make -j
sudo make install
```
## write a new cmake file
### set vars
```
# root dirs
set(CUPOCH_ROOT "/opt/cupoch/cupoch/")
# 增加对cupoch支持
find_package(PkgConfig QUIET)
find_package(pybind11  QUIET)
include(GenerateExportHeader)
if (PKGCONFIG_FOUND)
    pkg_search_module(EIGEN3          eigen3>=3.2.7   QUIET)
    pkg_search_module(GLFW            glfw3           QUIET)
    pkg_search_module(GLEW            glew            QUIET)
    pkg_search_module(JSONCPP         jsoncpp>=1.7.0  QUIET)
    pkg_search_module(PNG             libpng>=1.6.0   QUIET)
    pkg_search_module(JPEG_TURBO      libturbojpeg    QUIET)
endif (PKGCONFIG_FOUND)
set(CUPOCH_INCLUDE_DIRS ${CUPOCH_ROOT}/include
    ${CUPOCH_ROOT}/third_party
    ${CUPOCH_ROOT}/third_party/rmm/include
    ${CUPOCH_ROOT}/third_party/cnmem/include
    ${CUPOCH_ROOT}/third_party/fmt/include
    ${CUPOCH_ROOT}/third_party/liblzf/include
    ${PNG_INCLUDE_DIRS}
)
set(CUPOCH_LIBRARY_DIRS ${CUPOCH_ROOT}/lib)
set(CUPOCH_LIBRARIES
    cupoch_camera
    cupoch_collision
    cupoch_geometry
    cupoch_integration
    cupoch_io
    cupoch_odometry
    cupoch_planning
    cupoch_registration
    cupoch_utility
    cupoch_visualization
    tinyobjloader
    turbojpeg
    stdgpu
    flann_cuda_s
    fmt
    rmm
    liblzf
    rply
    ${CUDA_LIBRARIES}
    ${CUDA_CUBLAS_LIBRARIES}
    ${CUDA_curand_LIBRARY}
    ${PNG_LIBRARIES}
)
set(CUPOCH_NVCC_FLAGS
    -use_fast_math
    --expt-relaxed-constexpr
    --expt-extended-lambda
    --default-stream per-thread
    --use_fast_math
    -Xcudafe "--diag_suppress=integer_sign_change"
    -Xcudafe "--diag_suppress=partial_override"
    -Xcudafe "--diag_suppress=virtual_function_decl_hidden"
)
if(CMAKE_BUILD_TYPE STREQUAL "Debug" OR CMAKE_BUILD_TYPE STREQUAL "RelWithDebInfo" )
    list(APPEND CUPOCH_NVCC_FLAGS
        -G;-g
    )
endif()
set(CUPOCH_DEFINITIONS
    -DFLANN_USE_CUDA
    -DUSE_RMM
)
```
### using the cupoch
```
APPEND_TARGET_ARCH_FLAGS()

set(CUDA_NVCC_FLAGS ${CUDA_NVCC_FLAGS}
        ${CUPOCH_NVCC_FLAGS}
)
add_definitions(${CUPOCH_DEFINITIONS})
link_directories(${CUPOCH_LIBRARY_DIRS})
link_libraries(${CUPOCH_LIBRARIES})
add_executable(${EXE_NAME} src/${EXE_NAME}_node.cpp src/${EXE_NAME}.cpp)
```
## todo
You can write a FindCUPOCH.cmake as well

xiaopeige, 2021-01-29:
> # how to use the cupoch c++ library
> the following steps work in ubuntu1804
>
> ## install cupoch with 3rdparty libs
> add some install command to `third_party/CMakeLists.txt`
>
> ```
> # Installation [...]

> # how to use the cupoch c++ library
> the following steps work in ubuntu1804
>
> ## install cupoch with 3rdparty libs
> add some install command to `third_party/CMakeLists.txt`
>
> ```
> # Installation [...]

> # how to use the cupoch c++ library
> the following steps work in ubuntu1804
>
> ## install cupoch with 3rdparty libs
> add some install command to `third_party/CMakeLists.txt`
>
> ```
> # Installation [...]

thank you ,but I still can't use Cupoch directly, The error is that there is a problem with RMM. Do you have a similar problem?

ZhenshengLee, 2021-04-16:
> thank you ,but I still can't use Cupoch directly, The error is that there is a problem with RMM. Do you have a similar problem?

You could find it more in [perception_cupoch](https://github.com/ZhenshengLee/perception_cupoch)

Thanks.

ZhenshengLee, 2021-08-13:
@xiaopeige Hi, the compiling issue with RMM happens when you use cupoch that > 0.1.7, which upgrade thirdparty of rmm.

You could use [this fat source of cupoch](https://github.com/ZhenshengLee/cupoch-fat), which based on the version of 0.1.7

Thanks.

OrcunCanDeniz, 2021-10-13:
> # how to use the cupoch c++ library
> the following steps work in ubuntu1804
>
> ## install cupoch with 3rdparty libs
> add some install command to `third_party/CMakeLists.txt`
>
> ```
> # Installation [...]

Where should I put the new cmake file?

ZhenshengLee, 2022-12-14:
@dHofmeister your requirement is related with https://github.com/ZhenshengLee/perception_cupoch/issues/3

https://github.com/ZhenshengLee/perception_cupoch/tree/noetic#cupoch this will help to use cupoch with ros2 and cpp.

dHofmeister, 2022-12-15:
@ZhenshengLee Thank you. 

I will try and compile in an ubuntu 18.04 docker image.