neka-nat/cupoch #25: Add matcaps rendering to cupoch

1939938853, 2020-10-27:
Hello, 

Is this [matcaps](http://polyscope.run/features/materials/) rendering interesting to add to cupoch?

I think the biggest advantage for [matcaps rendering](https://www.google.com/search?source=hp&ei=rTaYX5T0Gc2LlwSQ3pagAQ&q=matcaps+rendering&oq=matcaps+rendering&gs_lcp=CgZwc3ktYWIQAzoFCAAQsQM6CAgAELEDEIMBOggILhCxAxCDAToCCAA6BQguELEDOgQIABAKOgQILhAKOgQIABAeOgYIABAKEB46BQghEKABOgYIABAWEB46CAgAEBYQChAeOggIIRAWEB0QHlD2I1jldmCie2gCcAB4AIABgwKIAdcKkgEGMTYuMS4xmAEAoAEBqgEHZ3dzLXdperABAA&sclient=psy-ab&ved=0ahUKEwjU_r7ChdXsAhXNxYUKHRCvBRQQ4dUDCAY&uact=5) is it is "cheap" to achieve and it looks very attractive!

Thank you!

neka-nat, 2020-10-28:
Sounds good!

1939938853, 2020-10-28:
[Here](http://www.beautybuddyapp.com/examples/matcap.html) is also a good matcaps implementation example.

closed 2021-02-21 by 1939938853