neka-nat/cupoch #27: Numpy conversion without going via. python list

saulzar, 2020-11-26:
Is there a way to convert a Vector3fVector to a numpy array without going via. a python list? This is much too slow for a large point cloud or image to be usable..

From the example on conversions:

```
print("---- To list of 1D np.array")
print(pcd.points.cpu())
print("---- To 2D np.array")
print(np.asarray(pcd.points.cpu()))
```

saulzar, 2020-11-26:
I came up with this for a point cloud, it works but also seems not very fast!
```

def to_cpu(pcd):
    from torch.utils.dlpack import from_dlpack

    points = from_dlpack(pcd.to_points_dlpack()).cpu().numpy()
    colours = from_dlpack(pcd.to_colors_dlpack()).cpu().numpy()
  
    cpu_cloud = open3d.geometry.PointCloud(open3d.utility.Vector3dVector(points))
    cpu_cloud.colors = open3d.utility.Vector3dVector(colours)
    return cpu_cloud

```

neka-nat, 2020-11-27:
Hi @saulzar, 

Thanks for testing.
As you say, I improved the handling of the conversion to numpy and transferring to host devices.
Please try using the master branch.

saulzar, 2020-11-27:
Hi @neka-nat!

That seems much better, thanks!

I also notice it is slower than I would expect to upload points from cpu... for example taking approximately 5 seconds to upload 10 million points, would this be expected?
```

>>> pcd.points = cph.utility.Vector3fVector(np.ones((1000000, 3)))          
```

I tried going via dlpack:

```

x = to_dlpack(torch.zeros(10000000).cuda())
pcd.from_points_dlpack(x)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: transform: failed to synchronize: cudaErrorIllegalAddress: an illegal memory access was encountered

```

neka-nat, 2020-11-27:
Copying from host to device has also been improved.
See below for a copy from pytorch.
https://github.com/neka-nat/cupoch/blob/master/examples/python/basic/from_torch_tensor.py

saulzar, 2020-11-27:
Much better now, great job!

Still get the cudaErrorIllegalAddress for from_points_dlpack, will make another issue about it though.

closed 2020-11-27 by saulzar