neka-nat/cupoch #32: Thirdparty

irenji, 2021-01-22:
Hello.

I decided to use cupoch in our project, since we needed to couple it with vcpkg. First idea was just to compile it seperatly.
But since cupoch have it's own dependencies it's conflicted with same dependecies in the vcpkg also in some cases like flann cupoch use minimal edited version.

So we decieded to start to change the build system of cupoch and make it possible to integrate it in the vcpkg.
I will be happy to share it here and improve it, currently it's not perfectly implemented but it's work for us.

neka-nat, 2021-01-22:
Hi,
Thanks for the great suggestion!
I agree that we should be able to use vcpkg.

In my opinion, I would like to modify the build system by making as many small changes as possible.
Is it possible to partially use vcpkg while keeping some of the original thirdparty?
Is it difficult to coexist with both?

irenji, 2021-01-22:
Actually, we can port the 3rdparties one by one. There will be no problem.

In normal cases, if you use cupoch in a small project there will be no problem. but linking it with project with having another version of those 3rdparties will cause linking problems.

But to start porting it we can do it one by one.

Also, I should mention that the flann that exists in the 3rdparty folder is not the official Flann. That needs to be fixed especially.

irenji, 2021-01-22:
Also, I should mention some of the 3rdparties are not available in vcpkg. we can start with the ones that is available.

neka-nat, 2021-01-25:
I agree to start porting them one by one!
If you submit pull requests, I will respond accordingly.
As for flann, it has some bug fixes in it, so it cannot replace the original one.

irenji, 2021-01-25:
Regarding Flann, Yes I saw your changes, I solved it by subclassing Flann (I can't remember the correct class name) in cupoch and integrate the changes in the subclass instead of changing Flann.

irenji, 2021-01-25:
Then I will close this pull request and start to port the 3rdparties one by one.

neka-nat, 2021-01-26:
I have not yet figured out what we should do about handling flann.
I also need to understand vcpkg better.
I don't foresee any problems with migrating common libraries such as glew.

closed 2024-11-06 by zellx3