neka-nat/cupoch #34: Use cupoch library independently

xiaopeige, 2021-01-29:
Hello, I want to compile Cupoch to generate the library on the windows system, but there is a memory manager allocation problem when calling it. Have you tried it on the window?

neka-nat, 2021-02-02:
Hi @xiaopeige ,
Thank you for reporting.
I have confirmed that it works on windows, but some problems have been reported. #30 
Is it possible to see the error messages?

xiaopeige, 2021-02-03:
for example ，it always have "unimplemented for this system" ,I don't know how to solve it

neka-nat, 2021-02-05:
What was the error you got when you ran it?
I found the following relevant issue.
https://stackoverflow.com/questions/58349473/thrust-static-assertion-when-using-in-cpp-files
Is it an error in cmake?

closed 2021-02-15 by neka-nat