neka-nat/cupoch #38: cluster_dbscan over voxel_down_sample function becomes slower in cupoch

AlexandrineRibeiro, 2021-02-08:
I test cluster_dbscan and voxel_down_sample  functions in both open3d and cupoch.  **I apply cluster_dbscan in the resulting point cloud from voxel_down_sample**  and cupoch is slower than open3d. 

I've the following processing times: 

For 196133 points:

```
geometry::PointCloud with 196133 points.
transform (CPU) [sec]: 0.0009989738464355469
transform (GPU) [sec]: 0.0009970664978027344
voxel_down_sample (CPU) [sec]: 0.011995553970336914
voxel_down_sample (GPU) [sec]: 0.01075434684753418
cluster_dbscan (CPU) [sec]: 0.05086064338684082
cluster_dbscan (GPU) [sec]: 0.16341233253479004
```

For 1440 points:

```
geometry::PointCloud with 1440 points.
transform (CPU) [sec]: 0.0
transform (GPU) [sec]: 0.0
voxel_down_sample (CPU) [sec]: 0.0
voxel_down_sample (GPU) [sec]: 0.001996755599975586
cluster_dbscan (CPU) [sec]: 0.002991914749145508
cluster_dbscan (GPU) [sec]: 0.9901258945465088
```

Is there anyway gpu could make the calculation faster than cpu?

neka-nat, 2021-02-11:
Hi,
Thank you for the question!
I have questions about your runtime environment.
* Is it a Windows environment?
* What is the number of OpenMP parallelism in Open3D?

AlexandrineRibeiro, 2021-02-11:
The environment tested on has the following specs:
-  Intel Core i7-8750H CPU
-  Nvidia GTX1070 GPU
-  OMP_NUM_THREAD=1
-  Python 3.7.9
-  Windows 10

neka-nat, 2021-02-11:
If you can use the ubuntu version, it may be faster.
The Windows version does not use a library called RMM, which uses a memory pool.
Also, these functions are not very fast, as you can see from the README results.

AlexandrineRibeiro, 2021-02-11:
Hi @neka-nat,

Thank you for your answer.

Feel free to close this issue.

closed 2021-02-11 by neka-nat