neka-nat/cupoch #39: cph.registration.registration_fast_based_on_feature_matching exits without error message

rcffc, 2021-02-09:
Console prints out:
:: Downsample with a voxel size 0.020.
:: Estimate normal with search radius 0.040.
:: Compute FPFH feature with search radius 0.100.
:: Downsample with a voxel size 0.020.
:: Estimate normal with search radius 0.040.
:: Compute FPFH feature with search radius 0.100.
:: Apply fast global registration with distance threshold 0.010

Process finished with exit code -1

rcffc, 2021-02-11:
Sometimes it would throw:
Process finished with exit code -1073740791 (0xC0000409)

rcffc, 2021-02-11:
I figured the error: 
```
cph.geometry.RGBDImage.create_from_color_and_depth(color_raw_cph, depth_raw_cph,
                                                                           depth_trunc=5.0,
                                                                           convert_rgb_to_intensity=False)
```

depth_trunc cannot be bigger than 5.0 otherwise I will get the error.

Why? I have no clue.

neka-nat, 2021-02-11:
Hi,
Thank you for reporting!
What is the difference in the number of points in the point cloud when depth_trunc is increased?

rcffc, 2021-02-11:
depth_trunc = 5.0 :   (1st PC) 647092, (2nd PC) 641854
depth_trunc = 4.99 : (1st PC) 424219, (2nd PC) 421553

rcffc, 2021-02-11:
I saw a spike to 100% in the GPU chart of my Task Manager. GPU out of memory seems to be the errror.

closed 2021-02-16 by neka-nat