neka-nat/cupoch #40: Do you have any plan to port cupoch for ROS2?

ldg810, 2021-02-15:
Hi, I am interesting with cupoch to use the library in Jetson Nano.

Do you have any plan to port the cupoch for ROS2 environment?

Thank you.

neka-nat, 2021-02-16:
Hi @ldg810 ,
Thanks for the question!
I've added an example for ros2.
This example receives pointcloud2 from [the realsense node](https://github.com/intel/ros2_intel_realsense/tree/refactor).
It works with the latest master.
https://github.com/neka-nat/cupoch/blob/master/examples/python/ros2/pointcloud2_msg.py

ldg810, 2021-02-18:
> Hi @ldg810 ,
> Thanks for the question!
> I've added an example for ros2.
> This example receives pointcloud2 from [the realsense node](https://github.com/intel/ros2_intel_realsense/tree/refactor).
> It [...]

Thank you very much!

closed 2021-02-18 by ldg810

ZhenshengLee, 2021-08-09:
@ldg810 Hi, you can use this repo https://github.com/ZhenshengLee/perception_cupoch/tree/foxy for cupoch + ros2 development, feel free to create issues.