neka-nat/cupoch #42: Out of memory while performing cph.registration.registration_filterreg

sujit-rai, 2021-02-17:
Hello,

I am using cph.registration.registration_filterreg for registering various pointcloud iteratively. The memory keeps growing after each iteration and after about 280 iterations i get  CUDA out of memory error.

Environment:

- Windows 10
- Python 3.7.9
- Intel Core i7-9700k
- NVIDIA GeForce GTX 1660, 6GB

[image]

neka-nat, 2021-02-19:
Thank you for the reporting!
I would like to investigate this bug.
Can you share your test code and data?

fredotran, 2021-03-20:
Hello,

I have the same issue as sujit-rai while I am using `cph.registration.compute_fpfh_feature` 
I am using a dataset with around 13 million points, and my test code is basically the same as yours.
[image]

neka-nat, 2021-03-26:
@fredotran 

Thanks!
I think this error is purely out of memory.
Try reducing the number of points.

closed 2021-04-07 by neka-nat

Agaskafara, 2024-11-05:
Hello,

Thank you so much for the repo, it is awesome.

I am using _cupoch.registration.registration_filterreg_ in the python wrapper and I think there is a memory leak (`MemoryError: std::bad_alloc: out_of_memory: CUDA error at: /home/runner/work/cupoch/cupoch/third_party/rmm/include/rmm/mr/device/cuda_memory_resource.hpp:70: cudaErrorMemoryAllocation out of memory`).

The following code prints the free gpu memory every time the function is called. Checked with nvidia-smi too.

Is there any way to free space available in the python wrapper? Is it really a memory-leak? Thanks

python: 3.10.15
cupoch: 0.2.11.0

```python
import numpy as np
import cupoch as cph
import cv2
import cupy as cp

def check_gpu_mem():
    free_bytes, total_bytes = cp.cuda.runtime.memGetInfo()
    free_mb = free_bytes / (1024 ** 2)
    total_mb = total_bytes / (1024 ** 2)
    return free_mb, total_mb

# Set rotation
transform = np.eye(4, dtype=np.float32)
transform[:3, :3] = cv2.Rodrigues(np.array([0.1, 0.2, 0.3]))[0]

a = np.random.random(10000*3).reshape(10000, 3).astype(np.float32)
a -= a.mean(0)
b = a.copy() @ transform[:3, :3]
src = cph.geometry.PointCloud(a)
tgt = cph.geometry.PointCloud(b)
criteria = cph.registration.FilterRegOption(max_iteration=50)
for i in range(30):
    out = cph.registration.registration_filterreg(src, tgt, criteria=criteria)
    free, total = check_gpu_mem()
    print(f"ITER {i+1}: [FREE:{free}]|[TOTAL:{total}]")
``` 
I got:

ITER 1: [FREE:4768.375]|[TOTAL:5695.1875]
ITER 2: [FREE:4586.375]|[TOTAL:5695.1875]
ITER 3: [FREE:4404.375]|[TOTAL:5695.1875]
ITER 4: [FREE:4222.375]|[TOTAL:5695.1875]
ITER 5: [FREE:4040.375]|[TOTAL:5695.1875]
ITER 6: [FREE:3858.375]|[TOTAL:5695.1875]
ITER 7: [FREE:3676.375]|[TOTAL:5695.1875]
ITER 8: [FREE:3494.375]|[TOTAL:5695.1875]
ITER 9: [FREE:3312.375]|[TOTAL:5695.1875]
ITER 10: [FREE:3130.375]|[TOTAL:5695.1875]
ITER 11: [FREE:2948.375]|[TOTAL:5695.1875]
...
ITER 21: [FREE:1040.125]|[TOTAL:5695.1875]
ITER 22: [FREE:852.125]|[TOTAL:5695.1875]
ITER 23: [FREE:666.125]|[TOTAL:5695.1875]
ITER 24: [FREE:480.125]|[TOTAL:5695.1875]
ITER 25: [FREE:292.125]|[TOTAL:5695.1875]
ITER 26: [FREE:106.125]|[TOTAL:5695.1875]
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
MemoryError: std::bad_alloc: out_of_memory: CUDA error at: /home/runner/work/cupoch/cupoch/third_party/rmm/include/rmm/mr/device/cuda_memory_resource.hpp:70: cudaErrorMemoryAllocation out of memory

stevedanomodolor, 2025-06-13:
@Agaskafara were you able to fix it in the end?