neka-nat/cupoch #48: Extracting point cloud from TSDF by frustum

VladMVLX, 2021-02-22:
Hello @neka-nat ,

Thank you for the great job you are doing with this project.

As I can observe from the code at current stage there is only full point cloud extraction possible from TSDF.
Are there any plans adding partial extraction by basic geometry figures such as frustum sphere or cube?

Thanks in advance.

neka-nat, 2021-02-25:
Hi,
Thank you for the question!
Mesh extraction is currently being worked on. #4
Do you need to extract primitive shapes instead of meshes?

VladMVLX, 2021-02-25:
Hello @neka-nat,

Thank you for the response.
What I mean is partial extraction of pointcloud / mesh from TSDF by bounding primitives.
Practically it could look like:
` TSDFVolume::ExtractPointCloud(const camera::PinholeCameraIntrinsic &intrinsic,  const Eigen::Matrix4f &extrinsic) ` - For frustum extraction

` TSDFVolume::ExtractPointCloud(const geometry::AxisAlignedBoundingBox &box) ` - For cube (AABB representation) extraction

or

` TSDFVolume::ExtractPointCloud(const geometry::Sphere &sphere) ` - For sphere extraction

Thus this functions will extract points which lays inside of geometry figure. 
This functionality can be very useful for realtime scanning and scan results editing.