neka-nat/cupoch #49: In windows use cupoch lib

xiaopeige, 2021-02-25:
I want use cupoch library in windows，like open3d，#include <Open3D/Open3D.h>, but there is a problem when using cupoch in winndows, is it supported now?

neka-nat, 2021-02-27:
Thank you for the question!
What kind of problem is it?

xiaopeige, 2021-03-01:
> Thank you for the question!
> What kind of problem is it?

Unable to reference third-party libraries, cupoch.lib cannot be used, I want to know if you have tested and used cupoch.lib on windows?

neka-nat, 2021-03-01:
I have confirmed that it works on windows, but there have been some problems reported. #30

xiaopeige, 2021-03-01:
> I have confirmed that it works on windows, but there have been some problems reported. #30

Yes, currently I can successfully generate the include folder and lib folder, but are there any examples of how to reference these two folders?

neka-nat, 2021-03-03:
The examples cmake will be helpful.
https://github.com/neka-nat/cupoch/blob/master/examples/cpp/CMakeLists.txt

closed 2021-03-06 by neka-nat