neka-nat/cupoch #50: Is there anyway to do multiple ICPs at once.

KalraA, 2021-03-07:
Hello @neka-nat,

This is an incredible project. Thank you so much! I have a quesiton:

I have N pairs of point clouds and I have to do ICP between each pair. I have 2 GPUs and I'm not sure if there is any good way of running simultaneous registration. I tried using python multprocessing but it actually slows it down by 3x rather than speeding it up. Any thoughts are welcome!

xiaopeige, 2021-03-08:
> Hello @neka-nat,
>
> This is an incredible project. Thank you so much! I have a quesiton:
>
> I have N pairs of point clouds and I have to do ICP between each pair. I have 2 GPUs and I'm not sure if there [...]

Excuse me, are you going to perform loop detection?

KalraA, 2021-03-08:
No, I'm trying to perform multiple object pose estimation. I have a point cloud of a scene with many objects, and a point cloud of each object. I'm trying to register each object point cloud to the corresponding part of the scene. I have an initial estimate for each object, and want to refine using ICP. 

Does this help?

neka-nat, 2021-03-14:
Hi,

Thank you for the question!
In the current implementation, the only way to do this is to execute them one by one in order.
In the future, it will be possible to speed up the process if ICP can be executed in multiple cuda streams.
In order to do so, it is necessary to implement ICP with specifying the cuda stream.

KalraA, 2021-03-16:
Hi @neka-nat, 

Thank you for your response. I am interested to try to implement this. Could you give me some pointers as to how you would go about doing it?

neka-nat, 2021-03-24:
The following site is helpful for handling CUDA stream.
Also, OpenCV provides stream interface in the functions, and parallelization is possible by explicitly specifying stream.
https://developer.download.nvidia.com/CUDA/training/StreamsAndConcurrencyWebinar.pdf
https://stackoverflow.com/questions/17842827/how-to-use-gpustream-in-opencv

closed 2021-04-09 by neka-nat

tomy807, 2025-01-06:
> No, I'm trying to perform multiple object pose estimation. I have a point cloud of a scene with many objects, and a point cloud of each object. I'm trying to register each object point cloud to the [...]

Hello, I’m also trying to solve the problem you were working on. May I ask how you resolved it? Although it has been a while, I’m leaving a comment hoping to get some advice.

tomy807, 2025-01-06:
> No, I'm trying to perform multiple object pose estimation. I have a point cloud of a scene with many objects, and a point cloud of each object. I'm trying to register each object point cloud to the [...]

Hi! @KalraA 
Hello, I’m also trying to solve the problem you were working on. 
May I ask how you resolved it? 
Although it has been a while, I’m leaving a comment hoping to get some advice.