neka-nat/cupoch #51: Fast global registration initial matching

Apeiria, 2021-03-10:
Hi @neka-nat ,

I'm learning your fast global registration code. I noticed that you are using brute force matching when matching FPFH features between src cloud and dst cloud to establish correspondence. Why aren't you using kdtree? Is build and searching kdtree slower than brute force matching on GPU?

Thanks in advance.

neka-nat, 2021-03-11:
Hi,

I use flann for the kdtree, but the kdtree implemented in flann is only for 3D data.
Therefore, I use brute force KNN for FPFH.
I haven't checked which one is superior in terms of speed, but
The computational complexity of the kdtree approaches that of the brute force as the dimensionality increases.

Apeiria, 2021-03-12:
> Hi,
> 
> I use flann for KDtree, but the kdtree implemented in flann is only for 3D data.
> Therefore, I use brute force KNN for FPFH.
> I haven't checked which one is superior in terms of speed, but
> The computational complexity of the kdtree approaches that of the brute force as the dimensionality increases.

Thank you for your reply!

closed 2021-03-12 by neka-nat