neka-nat/cupoch #52: cupoch v0.1.8 rmm and cuda 11.0 on Ubuntu

1939938853, 2021-03-11:
Hello,

It seems that cupoch v0.1.8 rmm crash with cuda 11.0 on Ubuntu?

neka-nat, 2021-03-14:
Hi,
What kind of crash is occurring?

1939938853, 2021-03-15:
Hi,

I use the master version, that can pass the Cuda 11.0 + RMM and Ubuntu 20.04 build.

But the performance with RMM seems not as my expected so significant. Can you give me any suggestion to improve the performance?

Thank you!

PS:

Here are some test

-- 1.9 million points VoxelDownSample to 60844 points
Open3D+OpenMP  65 ms
Cupoch+RMM  24 ms  

-- RemoveRadiusOutliers(100, 20.0), point_size = 60844
Open3D+OpenMP  114 ms
Cupoch+RMM  114 ms  

--Estimate Normals(), point_size = 60844
Open3D+OpenMP  59 ms
Cupoch+RMM  28 ms  

--Estimate Normals(), point_size = 23733
Open3D+OpenMP  21 ms
Cupoch+RMM  22 ms

neka-nat, 2021-03-15:
You may need to calculate more than 100,000 points to get the speed effect.
Also, using pool memory may make it a little faster.

1939938853, 2021-03-15:
How to use pool memory ?

neka-nat, 2021-03-18:
In python, you can do the following.
You can also use c++ in the same way.

```py
import cupoch as cph
cph.initialize_allocator(cph.PoolAllocation, 1000000000)
```

closed 2021-03-24 by neka-nat