neka-nat/cupoch #54: Compile failed on RTX3060

xiaopeige, 2021-03-17:
I compile cupoch on RTX3060 ，cuda11.0，win10，vs2019，cmake3.20.0，but compile failed.The error is as follows :
calling a __host__ function("__roundf") from a __device__ function("thrust::cuda_cub::__transform::unary_transform_f< ::thrust::detail::normal_iterator< ::thrust::device_ptr<float> > ,  ::thrust::detail::normal_iterator< ::thrust::device_ptr<unsigned long long> > ,  ::thrust::cuda_cub::__transform::no_stencil_tag,  ::,  ::thrust::cuda_cub::__transform::always_true_predicate> ::operator ()<long long> ") is not allowed	cupoch_geometry	C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.0\include\thrust\system\cuda\detail\transform.h	119	

identifier "__roundf" is undefined in device code	cupoch_geometry	C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.0\include\thrust\system\cuda\detail\transform.h	119	

“..\..\..\..\src\cupoch\geometry\trianglemesh.cu;..\..\..\..\src\cupoch\geometry\trianglemesh_factory.cu;..\..\..\..\src\cupoch\geometry\voxelgrid.cu;..\..\..\..\src\cupoch\geometry\voxelgrid_factory.cu”的自定义生成已退出，代码为 1。	cupoch_geometry	C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets	240	

I have tried for a long time and I don't know how to solve it. Thank you for your reply！

neka-nat, 2021-03-18:
Thanks!
I fixed in the latest master. Please give it a try.

xiaopeige, 2021-03-18:
> Thanks!
> I fixed in the latest master. Please give it a try.

Yes, I used this latest master to compile in the above environment, and the above error occurred

xiaopeige, 2021-03-18:
> Thanks!
> I fixed in the latest master. Please give it a try.

Sorry, just now I try compile again, success,Thank you!

closed 2021-03-18 by neka-nat