neka-nat/cupoch #55: Performance optimization on RTX3060

xiaopeige, 2021-03-19:
I saw you test this project on GTX1070 and cuda10.1. My current environment is RTX3060 and CUDA11.0. Is there room for optimization in the registration module, such as parameter modification and full utilization of GPU, thank you!

neka-nat, 2021-03-24:
Thanks!
There is no optimization capability for a specific architecture.
If you have any suggestions, please let me know.

xiaopeige, 2021-03-26:
> Thanks!
> There is no optimization capability for a specific architecture.
> If you have any suggestions, please let me know.

Thanks！
So does this performance have nothing to do with the graphics card? For example, the performance of using GTX1070 and RTX3060 is the same (in terms of time-consuming)?

neka-nat, 2021-03-26:
The performance is different.
My point was that the performance is not optimized for different architectures.
For example, I don't use the Tensor core available in the Volta generation.

xiaopeige, 2021-03-26:
Thanks！

I see that you are using the thrust library, so if use Tensor core, the performance of advanced graphics cards may be fully utilized, right?

neka-nat, 2021-03-26:
That's correct.
WMMA can be used to improve the performance of matrix multiplication.
https://developer.nvidia.com/blog/programming-tensor-cores-cuda-9/

closed 2021-03-27 by neka-nat