neka-nat/cupoch #56: KDTreeFlann

xiaopeige, 2021-03-29:
Hello, when I use EstimateNormals and ComputeFPFHFeature, I found that geometry::KDTreeFlann kdtree (that is, building kdtree) takes a lot of time. I want to update it here. Do you have any suggestions for improvement?

neka-nat, 2021-04-02:
I haven't tried it, so it may not work, but you could replace it with brute force nn.
https://github.com/neka-nat/cupoch/blob/master/src/cupoch/geometry/bruteforce_nn.h

xiaopeige, 2021-04-07:
Thank you, but I found that building kdtree in cupoch on gpu; and kdtree in open3d on cpu, the latter is faster, how should I explain this?

neka-nat, 2021-04-07:
Yes.
It depends on the number of points. the GPU is effective only when the number of points is more than 0.1~1 million.

xiaopeige, 2021-04-07:
My point count is always around 2000, so can change it to build kdtree on cpu?

neka-nat, 2021-04-07:
For that number, the cpu is the better choice. faster.

xiaopeige, 2021-04-07:
Yes, I want to use the cpu version of kdtree in Cupoch. Is this feasible?

neka-nat, 2021-04-07:
The cpu version is not implemented, so please use open3d or scipy's kdtree.

xiaopeige, 2021-04-07:
I put the kdtree of open3d in the 3rdparty of Cupoch, and then recompile, is this feasible?

neka-nat, 2021-04-07:
I'm sure you can.

xiaopeige, 2021-04-07:
ok ,thank you

closed 2021-04-07 by neka-nat

xiaopeige, 2021-04-09:
Build kdtree on cpu and gpu respectively. The difference lies in uploadtreetoGPU. This step takes more time, for a small number of point clouds (around 2000). I have completed the compilation of the cpu version of kdtree in Cupoch, but there is a problem with the operation, because the modules in the cupoch (such as EstimateNormals and ComputeFPFHFeature) are completed in the GPU, but the search kdtree is completed in the cpu， which is incompatible. Do you have any suggestions?

neka-nat, 2021-04-10:
If the number of points is about 2000, it would be faster to use the CPU for calculations other than kdtree.