neka-nat/cupoch #57: AttributeError: 'cupoch.cupoch.geometry.PointCloud' object has no attribute 'get_oriented_bounding_box'

SebasKoel, 2021-03-30:
Hi,

First of thanks for making this. Your work decrease the computation time significantly over standard Open3d.

I'm getting this error:
AttributeError: 'cupoch.cupoch.geometry.PointCloud' object has no attribute 'get_oriented_bounding_box'

This is my code:
`    plane_model, inliers = pcd.segment_plane(distance_threshold=0.03,
                                             ransac_n=50,
                                             num_iterations=100)
    inlier_cloud = pcd.select_by_index(inliers, invert=False)
    
    bbox = inlier_cloud.get_oriented_bounding_box()
    idx = bbox.get_point_indices_within_bounding_box(pcd.points)
    filtered_pcd = pcd.select_by_index(idx, invert=True)
`
Thanks

neka-nat, 2021-04-05:
Hi,

Thank you for the reporting!
I added get_oriented_bounding_box to the latest master. Please give it a try.

SebasKoel, 2021-04-06:
@neka-nat thank you! You're the best, bought you a coffee! I had some issues trying to make the build (I'm not the best in cmake, c++, or could it be that I'm on Ubuntu 20.04 and not 18.04?). Would it be possible for you to make a build and update PyPi?

neka-nat, 2021-04-06:
Thanks for the coffee!!
I updated the package, v0.1.9.1.
It's being built on CI and will be released in about an hour.
Please try to run pip when it's updated.

```
pip install -U cupoch
```

SebasKoel, 2021-04-08:
It works! Thanks a lot for the update.

closed 2021-04-08 by SebasKoel