neka-nat/cupoch #72: fatal error: cupoch/cupoch.h: No such file or directory

mvanlobensels, 2021-08-18:
I have installed Cupoch via pip as described in the docs. The Python bindings seems to work, however, when building the cpp examples I get the following error:

[image]

In the examples subfolder, I ran
```
mkdir build
cd build
cmake ..
make
```

This was tested on an Nvidia Jetson Nano

ZhenshengLee, 2021-08-18:
@neka-nat Time to consider the support to cpp development.

ZhenshengLee, 2021-08-18:
@mvanlobensels You can check this issue for more info about cpp development with cupoch. https://github.com/neka-nat/cupoch/issues/24

ZhenshengLee, 2021-08-18:
> Hi @ZhenshengLee, actually the reason I am asking this is because I get the same error when trying to build your perception-cupoch ROS package, when having installed cupoch via pip. But maybe I should ask this question there.

You need to install cupoch from source.

mvanlobensels, 2021-08-19:
When I install from source I get the following errors and it fails. I ran `cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/opt/cupoch/cupoch -DBUILD_UNIT_TESTS=OFF -DBUILD_PYBIND11=OFF -DBUILD_PYTHON_MODULE=OFF &&     make -j &&     sudo make install`, as descirbed in [#24](https://github.com/neka-nat/cupoch/issues/24)  which is executed during a docker build.

```
[ 41%] Completed 'ext_turbojpeg'
[ 41%] Built target ext_turbojpeg
[ 42%] Linking CXX static library ../../lib/libspdlog.a
[ 42%] Built target spdlog
[ 43%] Linking C static library ../../lib/libglew.a
[ 43%] Built target glew
[ 44%] Linking CXX static library ../../../lib/libcupoch_camera.a
[ 44%] Built target cupoch_camera
/tmp/cupoch/third_party/flann/algorithms/device_vector.h: In function 'decltype(auto) flann::exec_policy(cudaStream_t)':
/tmp/cupoch/third_party/flann/algorithms/device_vector.h:39:13: warning: 'rmm::exec_policy_t rmm::exec_policy(cudaStream_t)' is deprecated: Use new exec_policy in rmm/exec_policy.hpp [-Wdeprecated-declarations]
     return rmm::exec_policy(stream);
             ^~~~~~~~~~~
/tmp/cupoch/third_party/rmm/include/rmm/thrust_rmm_allocator.h:40:83: note: declared here
 [[deprecated("Use new exec_policy in rmm/exec_policy.hpp")]] inline exec_policy_t exec_policy(
                                                                                   ^~~~~~~~~~~
/tmp/cupoch/third_party/flann/algorithms/device_vector.h:39:31: warning: 'rmm::exec_policy_t rmm::exec_policy(cudaStream_t)' is deprecated: Use new exec_policy in rmm/exec_policy.hpp [-Wdeprecated-declarations]
     return rmm::exec_policy(stream);
                               ^
/tmp/cupoch/third_party/rmm/include/rmm/thrust_rmm_allocator.h:40:83: note: declared here
 [[deprecated("Use new exec_policy in rmm/exec_policy.hpp")]] inline exec_policy_t exec_policy(
                                                                                   ^~~~~~~~~~~
[ 45%] Linking CXX static library ../lib/libflann_cuda_s.a
[ 45%] Built target flann_cuda_s
[ 47%] Building NVCC (Device) object src/cupoch/utility/CMakeFiles/cupoch_utility.dir/cupoch_utility_generated_dl_converter.cu.o
[ 47%] Building NVCC (Device) object src/cupoch/utility/CMakeFiles/cupoch_utility.dir/cupoch_utility_generated_platform.cu.o
[ 47%] Building NVCC (Device) object src/cupoch/utility/CMakeFiles/cupoch_utility.dir/cupoch_utility_generated_eigen.cu.o
/tmp/cupoch/third_party/eigen/Eigen/src/Core/arch/Default/BFloat16.h(664): warning: calling a __host__ function from a __host__ __device__ function is not allowed

/tmp/cupoch/third_party/eigen/Eigen/src/Core/arch/Default/BFloat16.h(664): warning: calling a __host__ function from a __host__ __device__ function is not allowed

/tmp/cupoch/third_party/eigen/Eigen/src/Core/util/IntegralConstant.h(156): warning: calling a __host__ function from a __host__ __device__ function is not allowed
          detected during:
            instantiation of "Eigen::Index Eigen::internal::get_runtime_value(const T &) [with T=Eigen::internal::VariableAndFixedInt<-1>]" 
/tmp/cupoch/third_party/eigen/Eigen/src/Core/../plugins/BlockMethods.h(1261): here
            instantiation of "Eigen::DenseBase<Derived>::FixedSegmentReturnType<Eigen::internal::get_fixed_value<NType, -1>::value>::Type Eigen::DenseBase<Derived>::tail(NType) [with Derived=Eigen::Block<Eigen::Ref<Eigen::Matrix<float, 6, 6, 0, 6, 6>, 0, Eigen::OuterStride<-1>>, 6, 1, true>, NType=Eigen::internal::VariableAndFixedInt<-1>]" 
/tmp/cupoch/third_party/eigen/Eigen/src/LU/PartialPivLU.h(390): here
            instantiation of "Eigen::Index Eigen::internal::partial_lu_impl<Scalar, StorageOrder, PivIndex, SizeAtCompileTime>::unblocked_lu(Eigen::internal::partial_lu_impl<Scalar, StorageOrder, PivIndex, SizeAtCompileTime>::MatrixTypeRef &, PivIndex *, PivIndex &) [with Scalar=float, StorageOrder=0, PivIndex=int, SizeAtCompileTime=6]" 
/tmp/cupoch/third_party/eigen/Eigen/src/LU/PartialPivLU.h(439): here
            instantiation of "Eigen::Index Eigen::internal::partial_lu_impl<Scalar, StorageOrder, PivIndex, SizeAtCompileTime>::blocked_lu(Eigen::Index, Eigen::Index, Scalar *, Eigen::Index, PivIndex *, PivIndex &, Eigen::Index) [with Scalar=float, StorageOrder=0, PivIndex=int, SizeAtCompileTime=6]" 
/tmp/cupoch/third_party/eigen/Eigen/src/LU/PartialPivLU.h(510): here
            instantiation of "void Eigen::internal::partial_lu_inplace(MatrixType &, TranspositionType &, TranspositionType::StorageIndex &) [with MatrixType=Eigen::Matrix<float, 6, 6, 0, 6, 6>, TranspositionType=Eigen::Transpositions<6, 6, int>]" 
/tmp/cupoch/third_party/eigen/Eigen/src/LU/PartialPivLU.h(538): here
            [ 2 instantiation contexts not shown ]
            instantiation of "Eigen::PartialPivLU<_MatrixType>::PartialPivLU(const Eigen::EigenBase<InputType> &) [with _MatrixType=Eigen::Matrix<float, 6, 6, 0, 6, 6>, InputType=Eigen::Matrix<float, 6, 6, 0, 6, 6>]" 
/tmp/cupoch/third_party/eigen/Eigen/src/LU/PartialPivLU.h(599): here
            instantiation of "const Eigen::PartialPivLU<Eigen::MatrixBase<Derived>::PlainObject> Eigen::MatrixBase<Derived>::partialPivLu() const [with Derived=Eigen::Matrix<float, 6, 6, 0, 6, 6>]" 
/tmp/cupoch/third_party/eigen/Eigen/src/LU/Determinant.h(34): here
            instantiation of "Eigen::internal::traits<Derived>::Scalar Eigen::internal::determinant_impl<Derived, DeterminantType>::run(const Derived &) [with Derived=Eigen::Matrix<float, 6, 6, 0, 6, 6>, DeterminantType=6]" 
[... cut ...]
CMakeFiles/Makefile2:1423: recipe for target 'src/cupoch/utility/CMakeFiles/cupoch_utility.dir/all' failed
make[1]: *** [src/cupoch/utility/CMakeFiles/cupoch_utility.dir/all] Error 2
Makefile:135: recipe for target 'all' failed
make: *** [all] Error 2
The command '/bin/sh -c cd /tmp/cupoch && mkdir build && cd build &&     cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/opt/cupoch/cupoch -DBUILD_UNIT_TESTS=OFF -DBUILD_PYBIND11=OFF -DBUILD_PYTHON_MODULE=OFF &&     make -j &&     sudo make install' returned a non-zero code: 2
```

ZhenshengLee, 2021-08-19:
> Killed
> CMake Error at cupoch_utility_generated_eigen.cu.o.Release.cmake:280 (message):
>   Error generating file
>   /tmp/cupoch/build/src/cupoch/utility/CMakeFiles/cupoch_utility.dir//./cupoch_utility_generated_eigen.cu.o

It seems that the version of eigen in you container is outdated, try to upgrade the eigen.

mvanlobensels, 2021-08-25:
Thanks, that worked. I managed to build from source

closed 2021-08-25 by mvanlobensels