neka-nat/cupoch #73: How to run ROS node with cupoch

mvanlobensels, 2021-08-25:
I am trying to run a ROS node with cupoch on ROS Melodic, Ubuntu 18.04. However, since ROS Melodic uses Python2 and Cupoch only supports Python3, Cupoch is not found when running the ROS node via `rosrun`. When running the node directly via `python <ros node>`, cupoch is not found. When running the node directly via `python3 <ros node>`, ROS is not found. How am I supposed to use Cupoch with Python2 and ROS Melodic?

mvanlobensels, 2021-08-25:
I now see that cupoch 0.1.7.0 was the last version to support python2, but unfortunately it is not available anymore via pip

mvanlobensels, 2021-08-25:
In hindsight, 0.1.7.0 seems to be the last version to support python2, but the first version to support ROS IO was 0.1.8.0, which does not support python2 anymore.

mvanlobensels, 2021-09-01:
Hi @neka-nat,

Do you have a solution to this?

neka-nat, 2021-09-01:
Hi,
Thank you for the question!
By using virtualenv, it is possible to use python3 under melodic.
The following article is a good reference (in Japanese, but please follow the code).
https://qiita.com/otamasan/items/7ac7732a5c3d47ec3028

closed 2021-09-03 by neka-nat