neka-nat/cupoch #77: [error] [CreateToPointCloud2Msg] Width and Step sizes must be greater than 0

mvanlobensels, 2021-09-01:
Hi @neka-nat,

I have created the following simple ROS node to convert from a PointCloud2 to a Cupoch point cloud back to PointCloud2. This is based on your latest python example to convert from Cupoch to PointCloud2, however using the unchanged pointcloud does not work:

```
class PointCloudFilter:

    def __init__(self):
        self._point_cloud_sub = rospy.Subscriber('/camera1/depth/color/points', PointCloud2, self.point_cloud_callback)
        self._filtered_point_cloud_pub = rospy.Publisher('/camera1/depth/color/filtered', PointCloud2, queue_size=1)

    def point_cloud_callback(self, msg):
        pointcloud = cph.io.create_from_pointcloud2_msg(msg.data, cph.io.PointCloud2MsgInfo.default(msg.width, msg.point_step))

        data, info = cph.io.create_to_pointcloud2_msg(pointcloud)

if __name__ == '__main__':
    rospy.init_node('point_cloud_filter')
    filter = PointCloudFilter()
    rospy.spin()
```

However, this results in the following error:
`[error] [CreateToPointCloud2Msg] Width and Step sizes must be greater than 0.`

This is running on my laptop (x86_64), Python 3.6.9, CUDA 11.4. Thank you in advance.

neka-nat, 2021-09-01:
Thanks!
This is probably the same problem.
https://github.com/neka-nat/cupoch/issues/74
Please try to use the latest master.

mvanlobensels, 2021-09-01:
I suppose you mean this is not available via pip? So build from source? Is it possible for you to update the pip package?

neka-nat, 2021-09-01:
It needs to be built.
The pip package update will be coming soon.

mvanlobensels, 2021-09-01:
Thank you @neka-nat!

Unfortunately, I get build errors when building from source.
```
Killed
CMake Error at cupoch_geometry_generated_occupancygrid.cu.o.Release.cmake:280 (message):
  Error generating file
  /tmp/cupoch/build/src/cupoch/geometry/CMakeFiles/cupoch_geometry.dir//./cupoch_geometry_generated_occupancygrid.cu.o

```

neka-nat, 2021-09-04:
I released [v0.2.2.0](https://pypi.org/project/cupoch/).
Please try it!

closed 2021-09-06 by neka-nat