neka-nat/cupoch #79: Continous Mapping with D435 + T265

DieProgrammIDE, 2021-09-03:
Hello,
I am trying to programing a Robot with mapping.
I am using an NVIDIA Jetson Xavier NX, an Intel Realsense D435 Depth Camera and an Intel Realsense T265 Tracking Camera (Provides Position, Rotation as 4x4 Matrix). 

Both cameras are mounted to the Robot and move together with it.
I want to create a constantly updating map of the Environment, the real challenge to overcome is that the Robot changes its Location and Rotation constantly and that it needs to somehow be transformed so that it fits the previous Frames.

I can't figure out how to do this without having weird distortions.

Does somebody have an example code or an idea?

neka-nat, 2021-09-05:
Thank you for the question!
How about this document.
https://intel.github.io/robot_devkit_doc/pages/rs_slam.html

DieProgrammIDE, 2021-09-05:
Problem is my existing programm doesn't use ROS and i want to prevent importing to ROS, processing with ROS and exporting to my Programm again.

It would be good to find a solution in Python and without using ROS. The Realsense Examples are fairly close but they lack the transformation for T265.

neka-nat, 2021-09-18:
I have never used T265, but the following example may be helpful.
The following example gets the camera orientation from a file, but I think you can replace that with T265.
https://github.com/neka-nat/cupoch/blob/master/examples/python/advanced/rgbd_integration_uniform.py

closed 2021-09-21 by neka-nat