neka-nat/cupoch #82: [QST] How to compile a cpp file including a class containing device vector members

SpriteLin-ZJU, 2021-10-09:
Hello,
I am a rookie of gpu programming.
According to [,](https://github.com/NVIDIA/thrust/issues/526) ,
the device vector should not be included in the cpp file. But I see that the cpp file in your example folder calls a lot of classes containing device vector members. I want to know how this is compiled?
@neka-nat

Dcn303, 2024-02-06:
hi @neka-nat  by saying "it is compiled by nvcc and then called in the cpp file" did you meant that you have created a shared library file and then link that library file with the cpp file

neka-nat, 2024-02-06:
> hi @neka-nat by saying "it is compiled by nvcc and then called in the cpp file" did you meant that you have created that thrust program as a shared library file and then link that library file with the cpp file

You are correct!

Dcn303, 2024-02-06:
Thank you for your reply