neka-nat/cupoch #88: Occupancy grid not showing all points

tohmme, 2021-12-07:
Hi,

When having a point cloud and drawing this point cloud using `cph.visualization.draw_geometries([pcd])`, all points are succesfully drawn. However, when then trying to compute an occupancy grid from this given point cloud, not all points are shown on the grid. The considered points seem to be cut off at certain limit values. Is there anything in the source code that only allows to draw points in a certain region or discards certain points? I have looked at the source code and tried the example code but nothing seems to clarify what is wrong with my code.
Below is a part of the code that I try to run:
`import cupoch as cph`
`pcd = cph.geometry.PointCloud()`
`pcd.points = cph.utility.Vector3fVector(data)`
`ocg = cph.geometry.OccupancyGrid(1, 512)`
`ocg.insert(pcd, [0,0,0])`
`cph.visualization.draw_geometries([pcd])`
`cph.visualization.draw_geometries([ocg])`

using following data:
[data.csv](https://github.com/neka-nat/cupoch/files/7669335/data.csv)

Best regard and thanks in advance!

neka-nat, 2021-12-12:
[image]

neka-nat, 2021-12-31:
This has been fixed in master.
[image]

closed 2021-12-31 by neka-nat