neka-nat/cupoch #95: ICP algorithm have CUDA version?

xiaopeige, 2022-03-21:
I found that the ICP algorithm is not GPU accelerated. Does the ICP algorithm have a CUDA version?

neka-nat, 2022-03-21:
> I found that the ICP algorithm is not GPU accelerated. 

What does this mean?
All registration algorithms implemented in this library use CUDA.

xiaopeige, 2022-03-21:
Use ICP to find that it takes 10ms to construct 10,000 points of kdtree. I think this is relatively slow. Is there any way to improve it?

neka-nat, 2022-03-21:
You may need more points than that to benefit from the GPU.
The benchmark is tested with 1 million points, in which case it is about 100 times faster than the CPU.
https://raw.githubusercontent.com/neka-nat/cupoch/master/docs/_static/speedup.png

closed 2022-03-25 by neka-nat