neka-nat/cupoch #98: MemoryError cph.initialize_allocator(cph.PoolAllocation, 1000000000)

ducktrA, 2022-05-30:
All python examples complain about cph.PoolAllocation. Has anyone else seen that? cuda toolkit 11.6 
NVIDIA-SMI 510.60.02    Driver Version: 510.60.02    CUDA Version: 11.6  
There is plenty of GPU mem available. The 1G fits in easily. I can reduce the size heavily and same issue

traceback (most recent call last):
  File "basic/benchmarks3.py", line 5, in <module>
    cph.initialize_allocator(cph.PoolAllocation, 1000000000)
MemoryError: std::bad_alloc: RMM failure at:/home/adolf/cupoch/third_party/rmm/include/rmm/mr/device/pool_memory_resource.hpp:179: Maximum pool size exceeded

neka-nat, 2022-05-30:
Hi,
Does RMM work when used by itself?
https://docs.rapids.ai/api/rmm/stable/basics.html

```py
import rmm

pool = rmm.mr.PoolMemoryResource(
    rmm.mr.ManagedMemoryResource(),
    initial_pool_size=2**30,
    maximum_pool_size=2**32
)
rmm.mr.set_current_device_resource(pool)
```

neka-nat, 2022-06-01:
I have increased the version of RMM, so please try it with the latest master. (I don't use c++17, so it is not the latest RMM)

closed 2022-06-07 by neka-nat

ducktrA, 2022-06-20:
i did check out and built it. still the same issue. i remove my manually installed rmm.

ducktrA, 2022-06-20:
same on develop branch