neka-nat/cupoch #121: compile C++ static lib fail (master-v0.2.8.1), content of src/cupoch/visualization/shader/shader.h is wrong

six9326, 2023-06-29:
1. My operation: (Ubuntu 20.04）
    mkdir build && cd build
    cmake .. -DCMAKE_BUILD_TYPE=Release  -DBUILD_UNIT_TESTS=OFF -DBUILD_PYBIND11=OFF -DBUILD_PYTHON_MODULE=OFF -DBUILD_GLEW=ON -DBUILD_GLFW=ON -DBUILD_PNG=ON -DBUILD_JSONCPP=ON -DCMAKE_INSTALL_PREFIX=/home/cv_group/sdk/cupoch/master-0.2.8.1-static
    make -j6

2. Error log

[ 87%]  ……
[ 87%] Generating ../../../../src/cupoch/visualization/shader/shader.h
[ 87%] Built target shader_file_target
[ 87%] Building NVCC (Device) object src/cupoch/visualization/CMakeFiles/cupoch_visualization.dir/shader/cupoch_visualization_generated_image_shader.cu.o
/home/cv_group/tools/cupoch/cupoch-master-v0.2.8.1/src/cupoch/visualization/shader/shader.h(13): error: const variable "cupoch::visualization::glsl::cupoch" requires an initializer

I know shader.h is generated by scripts/encode_shader.py, but I don't know how to fix it.

4. vim cupoch-master-v0.2.8.1/src/cupoch/visualization/shader/shader.h ( wrong content)

  1 // Automatically generated header file for shader.
  2 // See LICENSE.txt for full license statement.
  3 
  4 #pragma once
  5 
  6 // clang-format off
  7 namespace cupoch {
  8 
  9 namespace visualization {
 10 
 11 namespace glsl {
 12 
 13 const char * const cupoch-master-v0 = R"(             //  Look at this line !!!,  variable name is wrong
 14 #version 330
 15 
 16 in vec2 UV;
 17 uniform sampler2D image_texture;
 18 
 19 out vec4 FragColor;
 20 
 21 void main()
 22 {
 23     FragColor = texture(image_texture, UV);
 24 }
 25 )";
 26 
 27 }  // namespace cupoch::glsl
 28 
 29 }  // namespace cupoch::visualization
 30 
 31 }  // namespace cupoch
 32 
 33 // clang-format on
 34 // clang-format off
 35 namespace cupoch {
 36 
 37 namespace visualization {
 38 
 39 namespace glsl {
 40 
 41 const char * const cupoch-master-v0 = R"(            //  Look at this line !!!,  variable name is wrong
 42 #version 330
 43 
 44 in vec2 UV;
 45 uniform sampler2D image_texture;
 46 
 47 uniform vec3 mask_color;
 48 uniform float mask_alpha;
 49 
 50 out vec4 FragColor;
 51 
 52 void main()
 53 {
 54     FragColor = vec4(mask_color, texture(image_texture, UV).r * mask_alpha);
 55 }
 56 

6. vim cupoch-branch-jetson-nano-v0.2.7/src/cupoch/visualization/shader/shader.h    ( correct content )

 1 // Automatically generated header file for shader.
  2 // See LICENSE.txt for full license statement.
  3 
  4 #pragma once
  5 
  6 // clang-format off
  7 namespace cupoch {
  8 
  9 namespace visualization {
 10 
 11 namespace glsl {
 12 
 13 const char * const image_fragment_shader = R"(             //  Look at this line !!!,  variable name is correct
 14 #version 330
 15 
 16 in vec2 UV;
 17 uniform sampler2D image_texture;
 18 
 19 out vec4 FragColor;
 20 
 21 void main()
 22 {
 23     FragColor = texture(image_texture, UV);
 24 }
 25 )";
 26 
 27 }  // namespace cupoch::glsl
 28 
 29 }  // namespace cupoch::visualization
 30 
 31 }  // namespace cupoch
 32 
 33 // clang-format on
 34 // clang-format off
 35 namespace cupoch {
 36 
 37 namespace visualization {
 38 
 39 namespace glsl {
 40 
 41 const char * const image_mask_fragment_shader = R"(             //  Look at this line !!!,  variable name is correct
 42 #version 330
 43 
 44 in vec2 UV;
 45 uniform sampler2D image_texture;
 46 
 47 uniform vec3 mask_color;
 48 uniform float mask_alpha;
 49 
 50 out vec4 FragColor;
 51 
 52 void main()
 53 {
 54     FragColor = vec4(mask_color, texture(image_texture, UV).r * mask_alpha);
 55 }

  
7. I want use cupoch in Nvidia Jetson Havier NX (16g),  Which branch is the most appropriate ?

neka-nat, 2023-06-30:
Looking at the generated shader.h, the file name is wrong.
Can you plant a debug print in the following CMakeLists and show me what you get?
https://github.com/neka-nat/cupoch/blob/master/src/cupoch/visualization/CMakeLists.txt

```
--- a/src/cupoch/visualization/CMakeLists.txt
+++ b/src/cupoch/visualization/CMakeLists.txt
@@ -1,4 +1,5 @@
 file(GLOB SHADER_FILES "shader/glsl/*.glsl")
+message("test******************** ${SHADER_FILES}")
 
 add_custom_command(
     OUTPUT ${PROJECT_SOURCE_DIR}/src/cupoch/visualization/shader/shader.h
diff --git a/src/python/make_install_pip_package.cmake b/src/python/make_install_pip_package.cmake
index b51d061..3358c66 100644
```

I obtained the following outputs.
```
test******************** /home/nekanat/project/cubpoc/src/cupoch/visualization/shader/glsl/image_fragment_shader.glsl;/home/nekanat/project/cubpoc/src/cupoch/visualization/shader/glsl/image_mask_fragment_shader.glsl;/home/nekanat/project/cubpoc/src/cupoch/visualization/shader/glsl/image_mask_vertex_shader.glsl;/home/nekanat/project/cubpoc/src/cupoch/visualization/shader/glsl/image_vertex_shader.glsl;/home/nekanat/project/cubpoc/src/cupoch/visualization/shader/glsl/normal_fragment_shader.glsl;/home/nekanat/project/cubpoc/src/cupoch/visualization/shader/glsl/normal_vertex_shader.glsl;/home/nekanat/project/cubpoc/src/cupoch/visualization/shader/glsl/phong_fragment_shader.glsl;/home/nekanat/project/cubpoc/src/cupoch/visualization/shader/glsl/phong_vertex_shader.glsl;/home/nekanat/project/cubpoc/src/cupoch/visualization/shader/glsl/picking_fragment_shader.glsl;/home/nekanat/project/cubpoc/src/cupoch/visualization/shader/glsl/picking_vertex_shader.glsl;/home/nekanat/project/cubpoc/src/cupoch/visualization/shader/glsl/rgmdimage_fragment_shader.glsl;/home/nekanat/project/cubpoc/src/cupoch/visualization/shader/glsl/simple2d_fragment_shader.glsl;/home/nekanat/project/cubpoc/src/cupoch/visualization/shader/glsl/simple2d_vertex_shader.glsl;/home/nekanat/project/cubpoc/src/cupoch/visualization/shader/glsl/simple_fragment_shader.glsl;/home/nekanat/project/cubpoc/src/cupoch/visualization/shader/glsl/simple_vertex_shader.glsl;/home/nekanat/project/cubpoc/src/cupoch/visualization/shader/glsl/simple_white_fragment_shader.glsl;/home/nekanat/project/cubpoc/src/cupoch/visualization/shader/glsl/simple_white_vertex_shader.glsl;/home/nekanat/project/cubpoc/src/cupoch/visualization/shader/glsl/texture_phong_fragment_shader.glsl;/home/nekanat/project/cubpoc/src/cupoch/visualization/shader/glsl/texture_phong_vertex_shader.glsl;/home/nekanat/project/cubpoc/src/cupoch/visualization/shader/glsl/texture_simple_fragment_shader.glsl;/home/nekanat/project/cubpoc/src/cupoch/visualization/shader/glsl/texture_simple_vertex_shader.glsl
```

neka-nat, 2023-06-30:
I have fixed it.
Try master.

six9326, 2023-07-01:
Compiled successfully !  Thanks very much !

closed 2023-07-01 by six9326